OpenCPN/OpenCPN #2900: S-52: Beacon sector thickness becomes very large in dual canvas mode

bc38, 2022-12-29:
Beacon sector thickness becomes very large in dual canvas mode. Appears if the canvases are divided proportionally and when one of the canvases is narrow
see screenshots.

It seems to me, but it would be better to slightly reduce the width of the sectors of lights in the normal state. Look at the visualization of S-52

[image]
[image]
[image]

Hakansv, 2022-12-29:
I see the same on Win. 
Both pure ENC and o-charts

bdbcat, 2022-12-29:
Corrected in github master now.  Not yet for o-charts...
Thanks

closed 2022-12-30 by bc38