OpenCPN/OpenCPN #2903: PI: Weather_routing  limits on Grib_pi version -needs update

rgleason, 2022-12-31:
In weather_routing_pi.cpp  about line 48
// Define minimum and maximum versions of the grib plugin supported
#define GRIB_MAX_MAJOR 4
#define GRIB_MAX_MINOR 1
#define GRIB_MIN_MAJOR 4
#define GRIB_MIN_MINOR 1

I cannot find the version number for grib_pi. Does it exist?

rgleason, 2022-12-31:
https://github.com/rgleason/weather_routing_pi/blob/master/src/weather_routing_pi.cpp

I looked in opencpn  grib_pi src and still cant find it.
If the version number no longer exists, what should be done about checking for it in wx_route?

rgleason, 2023-01-01:
Fixed for Climatology version, but have no version for grib_pi
https://github.com/OpenCPN/plugins/pull/782

leamas, 2023-01-01:
grib_pi is bundled in main opencpn so it has no independent, maintained version. That is, the very idea to check the grib_pi version is outdated since long.

If something like this should be checked it is the main opencpn version. IIRC this is not directly available for the plugin. However, API_VERSION_MINOR in _ocpn_plugin.h_  should be sufficient to figure out if main opencpn  is recent enough to have a usable grib_pi. 

I actually doubt that this needs to be checked at all, but might certainly be wrong. Havn't  checked the actual code.

Hakansv, 2023-01-01:
And when checked today wmm still reports major 4 and minor 1 while requested.
But I do agree about it would not be necessary to check the inbound wmm version.

leamas, 2023-01-01:
@Hakansv:  wmm? grib_pi?

Hakansv, 2023-01-01:
Hmmm. Yes sorry grib it would have been, not wmm. Thanks

rgleason, 2023-01-02:
User reports indicate wxroute + climatology now works.  Thank you.

closed 2023-01-02 by rgleason