OpenCPN/OpenCPN #2905: PI: Vdr_pi file not showing in nmea window ann Watchdog not operatiing

rgleason, 2023-01-01:
**Describe the bug**
VDR_pi running a nmea0183 file, does not work to show the stream of nmea0183 in the Nmea Message Window
As a result I think that watchdog_pi is not recieving any nmea, which makes it a appear that Watchdog is not working!

[image]

As I explained earlier, I think that OpenCPN has changed a lot, including the way all users expect to use VDR_pi.
Are we required to set something in connections to have the NMEA0183 message window work to show a NMEA data stream?

How are we to test many of these plugins without having this capability?

bdbcat, 2023-01-02:
Rick...
1. I am running vdr 1.3.26 on Win10, without problem.  The (very old) N0183 VDR stream is correctly shown in the NMEA Debug window.  Ownship follows the VDR contents.  AIS targets displayed.  All normal, just like O562.
2.  There is no need for anything new or special in  OCPN connections to enable debug window.  All is as it has been before in O562.
3. I do not know what your enabled serial port is all about.  But it is not normal to have an active serial port while playing a VDR recording.
4. Dont know about Watchdog. 
5. Please get VDR working correctly before adding any more complexity.  First rule of debugging:  Simplify to the minimum config that exhibits the problem you are studying.  Everything else is just noise.

rgleason, 2023-01-02:
OK, I removed the serial port, which I had done in VS2022 to try to get it going. That seems to get VDR and Nmea message working in normal Beta.O5.7.1  Then tried VS2022 without the serial port and it works with a data stream!
Turning on Watchdog, I found it immediately clamped down on the nmea message speed, so that it was one line at a time, and OpenCPN labored a lot and was very slow. Here is what alarms were on.  Pypilot Alarm is not even "ON" just configured.
  
[image]

Then I started deleting alarms the second or third one I deleted was Pypilot. Then Watchdog started acting properly and Opencpn ran better, nmea data flowed almost normally, ...some hickups and unevennesse.

Here are two shots running normally.
[image]

[image]

I have another question. In the attached debugging text. It says watchdog and other plugins don't have symbol definitions. How would I configure those and should I have them when debugging?

[watchdog-debug-output.txt](https://github.com/OpenCPN/OpenCPN/files/10329931/watchdog-debug-output.txt)

Thanks!

Also what should I do about the pypilot alarm?  It is not working and messes up Watchdog and OpenCPN.

rgleason, 2023-01-02:
BTW, I think that Message Screen in Nmea Message Window is a little too big. Is that some result of my having a high resolution screen?

bdbcat, 2023-01-02:
About PyPilot:
Speaking candidly: If it were my plugin to publish, right now, I would disable PyPilot watchdog functions.  Inform Sean that this has been done. Anyone complains, let's see if Sean has any interest in supporting Windows..
For full debugging of a plugin, you should copy the generated. ".pdb" file along with the ".dll" file, to whatever directory O loads the plugin from.

rgleason, 2023-01-02:
OK thats what I will try to do.   Sean's watchdog and mine have diverged somewhat too. Hopefully that can get fixed as well.
Thanks for the debug advice.  Closing

closed 2023-01-02 by rgleason