OpenCPN/OpenCPN #2907: Track properties dialog misses Time base selection

Hakansv, 2023-01-02:
**Describe the bug**
The row with selection of [ ] UTC, [ ] Local@PC and LMT@Track start is missing.

**To Reproduce**
Open preferences for a track.

**Expected behavior**
See attached picture.

**Desktop (please complete the following information if applicable):**
 - Win, Debian
 - Version [5.7.1]

**Screenshots**
[image]

Hakansv, 2023-01-09:
If related to the missing time set or not all track points are now set using UTC time. 
One would have expected local time.

bdbcat, 2023-01-10:
Corrected, wx32 related.

Hakansv, 2023-01-10:
Correction confirmed. Thanks

closed 2023-01-10 by Hakansv