OpenCPN/OpenCPN #2911: OCPN 5.7.1 requires OpenGL or it will not start

jongough, 2023-01-06:
**Describe the bug**
OCPN 5.7.1 requires OpenGL to be installed and active on windows or it will not start and crashes

**To Reproduce**
Steps to reproduce the behavior:
1. New install of windows 10 without OpenGL
2. Install OCPN
3. Run OCPN
4. Will display the main OCPN window and will then crash.

**Expected behavior**
OCPN should run but the OpenGL option in the control panel should be greyed out

**Screenshots**
None

**Desktop (please complete the following information if applicable):**
 - OS: MS Windows 
 - Version 10 & 11

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: N/A
 - Andorid version: [N/A

**Additional context**
It is not possible to run OpenGL on Windows running as a guest virtual machine. VirtualBox does partially run but screen issues occur and it can become unreadable. Linux QEMU/KVM does not support OpenGL at all.

bdbcat, 2023-01-06:
Working on that now.  Coming soon.

leamas, 2023-01-06:
Is indeed about OpenGL drivers. See  [#2912 comment](
https://github.com/OpenCPN/OpenCPN/issues/2912#issuecomment-1374226636) for work-around

bdbcat, 2023-01-07:
I spent some time working to get a clean build with OCPN_USE_GL:OFF.
Finally concluded that it simply may not be possible without significant effort.  The key problem is our private class LLRegion().  This class is used to calculate unions and intersections of geographical areas on screen, critical to the quilting process.  LLRegion uses GLU internally, as we have found that the GLU tessellator is by far the fastest method to perform the geometric logic required.  And GLU requires basic GL headers and core functionality.  Note that this requirement has nothing to do with actual rendering operations on-screen.  The operations required are abstract logic paths.  Sigh...
I have not given up yet, but I would not pin my hopes on seeing this change anytime soon.

jongough, 2023-01-13:
I have managed to get a build of OCPN running on Windows 10 on a non-OpenGL qemu/kvm virtual client. To get it running I put a hard `return false;` into OCPNPlatform::IsGLCapable() straight after the test for android and CLI. This may not work in all cases, but perhaps there could be a flag, for testing at least, to allow this to run. I think there should be a better solution, but....

bdbcat, 2023-01-13:
Jon...
That is indeed interesting.  If that is all that was needed to get it building and running, then there must be some OpenGL libraries/headers available on the system.  They just don't run correctly, I suppose.
Thoughts on that?

jongough, 2023-01-13:
I am sure the header files are all OK, it is only when a function tries to use the OpenGL functions will it come to grief. For me, once I found the issue, it was quite simple to work around it to diagnose the issue with OD. However, I think a real solution needs to be implemented such that if the opencpn.ini/conf file says not to use OpenGL then it should not be used.

OD makes use of tessellation functions for some of its drawings but it will work on non OpenGL systems quite happily. Is it possible to find a subset that you need for speed without having to actually use the real graphics functions for the drawing?

jongough, 2023-02-23:
Here is the frame stack when it fails. The failure occurs where the test of opengl capability passes, but it fails when this capability is used:
```
Exception thrown at 0x00D92022 in opencpn.exe: 0xC0000005: Access violation reading location 0x00000000.
 	
 	>	[Inline Frame] opencpn.exe!std::_Narrow_char_traits<char,int>::length(const char * const) Line 418	C++
 	[Inline Frame] opencpn.exe!std::string::{ctor}(const char * const _Ptr) Line 2674	C++
 	opencpn.exe!OCPNPlatform::BuildGLCaps(void * pbuf) Line 728	C++
 	opencpn.exe!OCPNPlatform::IsGLCapable() Line 795	C++
 	opencpn.exe!OCPNPlatform::Initialize_3() Line 641	C++
 	opencpn.exe!MyApp::OnInit() Line 1728	C++
 	[External Code]	
 	[Frames below may be incorrect and/or missing, no symbols loaded for wxbase32u_vc14x.dll]	
 	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 953	C++
```
I am using a workaround to allow debugging in windows 10, but this is not suitable for a production version.

bdbcat, 2023-02-23:
Thanks for the trace.  Should be possible to protect against this crash.
I'll push a patch to test.

jongough, 2023-02-23:
My current workaround is to modify OCPNPlatform.cpp at line 788:
```
bool OCPNPlatform::IsGLCapable() {
#ifdef __OCPN__ANDROID__
  return true;
#elif defined(CLI)
  return false;
#else
  return false;
```
with the `return false;` which forces it not to try using opengl.

leamas, 2023-02-23:
I would call that a bugfix rather than a work around. In no cases a CLI build is GL capable, and OCPNPlatform is the place where we should handle these kind of differences. 

In other words: returning true in a CLI build is just always wrong.

bdbcat, 2023-02-23:
Jon...
I pushed a patch which should avoid the fault reported.
I have no way to test with the virtual environment you are using, so I would appreciate it if you build and test from master, and report results.
Thanks

jongough, 2023-02-23:
I have just tried it and it seems to be working. Thanks.

closed 2023-02-23 by jongough