OpenCPN/OpenCPN #2912: OCPN 5.7.1 Visual Studio 2022 build process documentation does not work

jongough, 2023-01-06:
**Describe the bug**
Trying to build OCPN 5.7.1 following the documentation does not work. The build process fails at many critical points

**To Reproduce**
Steps to reproduce the behavior:
1. Follow instructions found here:  https://opencpn-manuals.github.io/main/ocpn-dev-manual/0.1/od-compile-windows.html
2. Link to this page is difficult to find and often ends in circular html not available errors
3. Explanation of using the process is incomplete leaving The user at unhelpful error messages
4. 

**Expected behavior**
A clear and concise description of the steps needed to get a windows development environment installed and working
 
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: MS Windows
 - Version 10 & 11

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: N/A
 - Andorid version: N/A

**Additional context**
This process needs to work to enable debugging of issues found. There is a need to be able to step through the running version of OCPN and any plugins that are causing issues. This is hindering the resolution of an issue of icons within a plugin.
So far the install process assumes the user has an intimate knowledge of MS Windows and Visual Studio and is running this on a dedicated windows machine. If this is not the case then the process is almost impossible to follow.

So far the following extra packages need to be hand installed:

1. cmake
2. 7-zip
3. python

Administrator powershell needs extra knowledge to get working as it spawns a new environment in which network mappings do not exist, changes do not persist and commands have a different syntax to the standard Developer Command Prompt.

The following command is not run to set the wxWidgets environment:

> cache\wx-config.bat

leamas, 2023-01-06:
Also note that the the description in the manualdoes not cover setting up the complete VS environment, it is just about making a CLI build. I do understand the need for such a description, but does nto use VS myself so I'm certainly not the right person to do that.

Patches  to the manual welcome!

leamas, 2023-01-06:
> So far the following extra packages need to be hand installed:

> > All these, as well as some other build-deps, are installed by buildwin\win_deps.bat

No, you are right, they are not all installed by win_deps.bat. Seems that they was lost while adapting to the CI build environments which have them installed by default.

Anyway, this is then a bug which needs to be fixed.

jongough, 2023-01-06:
I don't use windows for anything apart from trying to test issues that are raised. In this case I need to be able to build OPCN and get it running under VS2022 so I can debug what is happening. I can build/use the appveyor built version, but it just crashes on startup on a windows machine without OpenGL installed. So I need to get past that. Before this release it was fairly simple to get a windows build working on VS2017 (and previous), but Windows, VS, the build process and OCPN have all changed and it is not clear how to build a testing version. Hence raising the issue hoping that some windows guru can point me, and others, in the correct direction.

jongough, 2023-01-06:
I can get virtualbox win10 OPCN running but cannot build on it. I can get qemu/kvm win10 OCPN to build but not run. 

As for the crash, there is nothing to report, there is nothing in the OpenCPN.log, there is nothing on the console when run from it. It seems to crash immediately after drawing the first panel (main screen of OCPN) but before anything is drawn on it. If I can get a debug build built that VS2022 will work with I can give more information. But....

leamas, 2023-01-06:
I can build on VirtualBox, so something is different in your setup. Using VBox 7.0.4 on Fedora 38.  Guest settings:  7GB memory, 4 processors.

I can reproduce your crash direct after starting O.

leamas, 2023-01-06:
The crash seems, unsurprisingly, be about the opengl drivers.  Found  a  [forum entry](https://forums.virtualbox.org/viewtopic.php?t=101304#p491844).  Downloaded the _opengl32.dll_ mentioned there and copied it to _\Program Files (x86)\OpenCPN_  -> no to crash on startup.

This is just a work around for now, using mesa sw-emulated OpenGL is most likely not a long term solution.

leamas, 2023-01-06:
The crash on startup is worth a separate issue. I think I saw one. Did I?

jongough, 2023-01-06:
Yes, #2911

nohal, 2024-02-14:
I have followed the current documentation multiple times om clean Windows machines and see no problems.