OpenCPN/OpenCPN #2916: Win10 build process does not set path correctly for gettext

jongough, 2023-01-07:
**Describe the bug**
Building on Windows10 using the new, 5.7.1 build process does not set the path statement correctly when installing poedit (the installation that provides the gettext required packages. This causes the  ci\appveyor.bat file build process to fail. The cmake process gives the following:
```
CMake Error at C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Gettext (missing: GETTEXT_MSGMERGE_EXECUTABLE
  GETTEXT_MSGFMT_EXECUTABLE)
Call Stack (most recent call first):
  C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.24/Modules/FindGettext.cmake:81 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:1238 (find_package)

-- Configuring incomplete, errors occurred!
```

**To Reproduce**
Steps to reproduce the behavior:
1. Follow build instructions on webpage: `https://opencpn-manuals.github.io/main/ocpn-dev-manual/0.1/od-compile-windows.html`
2. Open VS2022 developer prompt as administrator
3. exectute buildwin\win-deps.bat
4. execute ci\appveyor.bat
5. error occurs

**Expected behavior**
Expect vs2022 to run cmake and then build OCPN into the 'build' directory

**Screenshots**
N/A

**Desktop (please complete the following information if applicable):**
 - OS: Windows 10 running as guest on VirtualBox
 - Version: Windows 10, VirtualBox 7.0.4

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: N/A
 - Andorid version: N/A

**Additional context**
Trying to build a runnable and debuggable version of OCPN to investigate issues on Windows with icon sizes and svg files.

Current workaround is to modify the path statement by adding `C:\Program Files (x86)\Poedit\Gettexttools\bin;` to the current cmd prompt path statement.

leamas, 2023-01-07:
hm... just to analyze: Same error if you run from a plain cmd prompt i. e., not from a  VS2022 developer prompt?

Note that the build is best performed without administrative privileges after the first  win_deps.bat run.

jongough, 2023-01-07:
I ran the first pass on a new git clone of OpenCPN in an administor dev prompt, as per the instructions, to get the software installed. Then ran from a straight command prompt (cmd in the start search bar). This gave the error about not running with admin privileges for choco install of poedit, then gave the cmake error about not finding the package.
I did restart the cmd prompt, but I did not reboot the machine, to make sure it picked up the latest changes to environment variables.
Without admin privileges there is always an error when running ci\appveyor.bat:
```
C:\Users\jon\OpenCPN>ci\appveyor.bat

C:\Users\jon\OpenCPN>setlocal enabledelayedexpansion

C:\Users\jon\OpenCPN>msgmerge --version  1>nul 2>&1

C:\Users\jon\OpenCPN>if errorlevel 1 (
choco install -y poedit
 set "PATH=C:\Python311\Scripts\;C:\Python311\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\ProgramData\chocolatey\bin;C:\Program Files\Git\cmd;C:\Users\jon\AppData\Local\Microsoft\WindowsApps;;C:\Program Files (x86)\Poedit\Gettexttools\bin"
)
Chocolatey v1.2.1
Chocolatey detected you are not running from an elevated command shell
 (cmd/powershell).

 You may experience errors - many functions/packages
 require admin rights. Only advanced users should run choco w/out an
 elevated shell. When you open the command shell, you should ensure
 that you do so with "Run as Administrator" selected. If you are
 attempting to use Chocolatey in a non-administrator setting, you
 must select a different location other than the default install
 location. See
 https://docs.chocolatey.org/en-us/choco/setup#non-administrative-install
 for details.

For the question below, you have 20 seconds to make a selection.

 Do you want to continue?([Y]es/[N]o): y

Installing the following packages:
poedit
By installing, you accept licenses for the packages.
poedit v2.4.2 already installed.
 Use --force to reinstall, specify a version to install, or try upgrade.

Chocolatey installed 0/1 packages.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
```
I haven't raised an issue on this, though it should be fixed, but it, currently, is not critical just confusing.

jongough, 2023-01-13:
It would appear that the build process is currently aimed at a batch build in appveyor and is not intended for debugging purposes. There are changes that need to be made to the underlying environment to be able to use VS2022 GUI. The appveyor processes creates a temporary environment suitable for a build, possibly, but relies on good internet and being successfully setup.

I need to debug OCPN and a plugin which requires the use of the VS2022 GUI and the ability to build in the GUI so that single stepping through the code is possible along with the ability to investigate variable values. To get this to work there is  need to:

1. Install wxWidgetss 3.2.1 locally
2. Set the environment variables:
     - WX_DIR,
     - wxWidgets_INCLUDE_DIRS, 
     - wxWidgets_LIB_DIR, 
     - wxWidgets_LIBRARIES, 
     - wxwidgets_Root-DIR, and 
     - WXWIN 
     - to point to the local install of wxwidgets.
3. Add paths to the path variable for:
     - GettextTools
     - Git
     - chocolatey
     - cmake
     - 7-Zip
     - wxWidgets dll directory
4. May have to add buildwin to the path statement
5. If other files are not found add their paths to the path statement as well.

Some of these executables will need to be added to the system by running the install process found in both `buildwin\win_deps.bat` and `ci\appveyor.bat`

There may be a need to run the appveyor.bat process, but anything that installs any of the above or creates 'different' values for the variables above will need to be commented out with the `::` comment pattern.

If you are running windows with OpenGL installed it should now be possible to build and run OpenCPN. If you don't have or cannot have OpenGL installed on your windows the runtime will fail.

bdbcat, 2023-01-13:
Jon...
Are you stuck on this, or simply commenting for the record?
I agree that the CI build does not work very well locally.
How can I help get you debugging now?  Rick and Hakan are doing so, after some fiddling.
Dave

jongough, 2023-01-13:
I have been stuck on many small items trying to get OCPN to run on Windows 10 virtual client. I 'think' I may have it running now by bypassing OpenGL (see issue #2911). I can now try to see what the issue with OD is. However, I am about to go out on the boat for a couple of weeks so may not get to this issue until then.

The above about the setup for VS GUI was to try an help others get past the new build process so that debugging can occur. It has taken me nearly two weeks of thinking, trying and failing to get my head around windows, the 'new' build process for windows and VS2022to get this far. I also seem to be able to run simplistic tests, so I hope this is going to be enough to identify the icon issue.

nohal, 2024-02-14:
I have used the current documentation and build scripts on clean Windows installations and did not encounter any problems.

closed 2024-02-14 by nohal

rgleason, 2024-02-14:
The link page for Developer Manual has been updated sometime ago.
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:dev_manual

at the bottom of this page are two links for building Windows which TransmitterDan and Alec worked on, and which I find to  be very useful.    I use the second link if I am going to build for plugins and test
[Complete builds using Visual Studio 2022 Community Edition or Visual Studio 2019 Community Edition](https://opencpn-manuals.github.io/main/opencpn-dev/windows.html#_complete_builds_using_visual_studio_2022_community_edition_or_visual_studio_2019_community_edition)

Perhaps Jon is following some other older page?