OpenCPN/OpenCPN #2922: OpenCPN function LoadSVG does not work if a max size is given

jongough, 2023-01-13:
**Describe the bug**
ocpn_draw_pi (OD) loads its icons to use for its own floating toolbar by using the OCPN provided GetBitmapFromSVGFile function in pluginmanager.cpp. This uses the LoadSVG function in svg_utils.cpp. With the introduction of wxWidgets 3.2.1 a new function is being used to load the svg icons, namely `wxBitmapBundle::FromSVGFile`. This function takes the size to return the loaded bitmap  from the svg. OD only provides the maximum size for the platform uint by the use of `-1` although `MAX_UINT` will give the same size.

It would appear that the use of this function does not work with this allowed size. OCPN knows what size icons it is using or its toolbar would also not work. With the use of the maximum size the bitmap is not loaded and the error message given by OD is: `Failed to load all OCPN Draw Plugin icons, check OCPN log for details`.

**To Reproduce**
Steps to reproduce the behavior:
1. Run OCPN on windows 10
2. Load/install/activate OD
3. The message `Failed to load all OCPN Draw Plugin icons, check OCPN log for details` will be given
4. OD works OK, but the floating toolbar will not have any displayable icons.

**Expected behavior**
The size process for icons has not changed for some time and it appears to work on all platforms apart from MS Windows. With the change to wxWidgets 3.2.1 this appears to have changed

**Screenshots**
Not needed, just run OD on a windows platform with OCPN 5.7.1

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Version 10

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: N/A
 - Andorid version:N/A

**Additional context**
None

rgleason, 2023-01-13:
See for TestPlugin https://github.com/jongough/testplugin_pi/issues/280
for Ocpn_draw  https://github.com/jongough/ocpn_draw_pi/issues/513

leamas, 2023-01-13:
Perhaps it's about wx32, perhaps not. LoadSVG is basically just a shim for libwxsvg which is a separate library. For many platfoem we are bundling this, and the version bundled IIRC really old. Might make sense to update.

rgleason, 2023-01-13:
So all windows plugins get to convert back to bitmaps just because nobody bothered to update libwxsvg for windows, but did update it for ubuntu linux and mac?

leamas, 2023-01-13:
Debian has libwxsvg packaged, maintained by you know who. For others it would be about updating the sources in libs/wxsvg from https://git.code.sf.net/p/wxsvg/git

jongough, 2023-01-14:
It also looks like there is a conflict in definition of width and height. the LoadSVG function in svg_utils.cpp has both width and height defined as unsigned int, however wxBitmapBundle has the size specified as int. The default size, wxDefaultSize is int where it uses the value -1 to say use the default size. This is probably the root of the issue with OD as it is saying use the default size but it has not been set. Also the conflicting definitions of size should really be sorted out or handled within the code (another method with signed integers?).

bdbcat, 2023-01-14:
Jon...
I just tested on VS2022, wxWidgets 321 built in debug mode.
Despite the type definition of the argument, wxBitmapBundle::GetBitmap(s)  returns an invalid bitmap if size input parameter is <= 0.  We might expect it to honor the pixel size set in the SVG file, but it simply does not, at least on Windows.  Full stop.

The correct solution here is to call for bitmaps of an appropriate dimension for the application.
I do not know how OD plugin handles tool icon sizes for different display densities.  Perhaps they are simply scaled from some  "default" size to whatever is required on screen.  This is sub-optimal, from appearance point.  Better to request the bitmap at the target final resolution to preserve the image definition present in the SVG.  That is, after all, the only real reason to use Scaled-Vector-Graphics in the first place, since they are slower to decode than things like PNGs..

Also, some misinformation here.  wxBitmapBundle does not use wxsvg at all.  It uses an internal SVG decoder( a port of NanoSVG).

bdbcat, 2023-01-14:
More...
It is quite possible that a size <0 parameter works on wx30, which may be what your current linux build links to.  In this case, wxBitmapBundle is not used, and the wxsvg library is used.  Maybe it has  different behavior with size < 0 input.
O571 for Windows and MacOS definitely use wx32, and so use wxBitmapBundle.

rgleason, 2023-01-14:
This does not seem to work  https://git.code.sf.net/p/wxsvg/git

jongough, 2023-01-14:
OD loads its icons by calling GetBitmapFromSVGFile in pluginmanager. The default size is set, in this case -1, to let OCPN provide the size in its loading of the svg file. Is there an OCPN plugin API that will provide the default size of icons to use on the currently running environment? I not, perhaps one should be provided as OCPN already knows the default size of icons to use as it has already populated the main toolbar.

bdbcat, 2023-01-14:
That is an interesting approach.  If you knew the size of, say, a main main toolbar icon, would that be enough.? You could scale up or down slightly from that, as necessary.

bdbcat, 2023-01-14:
Or, using some code from Dashboard:

  // Scale the images in the dashboard list control
  int imageRefSize = 32 * GetOCPNGUIToolScaleFactor_PlugIn();
  bmDashBoard = GetBitmapFromSVGFile(filename, imageRefSize, imageRefSize);

That should provide a reasonably consistent look.

jongough, 2023-01-14:
Yes, that would help.
What is the `32 *  ` for?
Is this approach consistent on all resolutions?

jongough, 2023-01-14:
However, isn't that really two calls to sort out a problem which could be done within the OCPN loadsvg based on a default setting, -1, or a specified non-default setting?

bdbcat, 2023-01-14:
1.  32 is the base size, if scale factor is unity.  And that gives useful icon sizes on Dashboard.
2. Yes, it is two calls.  But you need that to obtain the user preferences for GUI scale factor.
3. Really, we do not want OPCN core guessing at the desired bitmap size on default values.  There is no default bitmap size as rendered from SVG.  That is the whole point of SVG.

jongough, 2023-01-14:
Is it possible for the end user to change the display resolution without restarting OCPN and if so is there a notification to the plugins of this to get the icons resized?

bdbcat, 2023-01-14:
Is it possible for the end user to change the display resolution without restarting OCPN?
No.  Display parameters are established on OCPN startup.

jongough, 2023-01-14:
OK. I have updated ODicons and will try and push it up soon. I think I will have to change the properties dialog to allow the user to decided the size for themsleves, but that will have to wait a couple of weeks.

bdbcat, 2023-01-14:
User selectable sizing would be great.  With all the variations in pixel density, and external scaling factors across the various platforms, we really do need lots of control points for the user.

jongough, 2023-01-14:
I have just uploaded OD 1.8.20.0 into the beta catalog awaiting a merge that seems, on my windows system, to display the OD toolbar correctly with icons. It will need to be checked on high definition OpenGL on Win 10 & 11, as I don't have that type of environment.

bdbcat, 2023-01-14:
Thanks, will do.

rgleason, 2023-01-14:
Jon please see https://github.com/jongough/testplugin_pi/issues/281#issuecomment-1382930092   big progress has been made!

nohal, 2024-02-25:
Closing, plugin is responsible for knowing how big it wants it's assets. I will make the newly available monitor info available in API 1.19, but in case the plugin owns any windows, it still needs to discover on which monitor they currently are in the `wxMoveEvent` handler and do whatever is appropriate for it.

closed 2024-02-25 by nohal