OpenCPN/OpenCPN #2929: Question: Why would an RPI plugin for 5.7.1 load with 5.6.2?

rgleason, 2023-01-14:
**Describe the bug**
I am not sure this is a bug, but it is a question that is puzzling.
See this Issue in autopilot_route, where Marco Bergman is testing the 0.4.8.0 plugin
https://github.com/marcobergman/ais_simulation/issues/10#issuecomment-1382717747
https://github.com/marcobergman/ais_simulation/issues/10#issuecomment-1382732936

RPI Opencpn 5.6.2 loaded a 5.7.1 plugin in Beta catalog "autopilot_route 0.4.8.0"

Why?

bdbcat, 2023-01-14:
No mystery there. RPi Bullseye uses wx31, so an older plugin will load just fine.
Would not work on Windows or Mac, though.

rgleason, 2023-01-14:
Thanks for that quick answer Dave. 
Marco Bergman is trying to build RPI and also trying Ubuntu builds.
He is getting this failure
"I get the same message on Ubuntu. Now I give up ;-)
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS)"

What would you suggest he do to fix this? Is there some command to install wxWidgets or to direct the env to the wxWidgets location?

He wants to build autopilot_route 0.4.8.0 and be able to debug it and step through.

Hakansv, 2023-01-14:
Not an answer but to share my experiences:
On my RPi: After having installed dependencies as advised in the INSTALL file in O repo it works to build both OCPN and some plugins. ( (A small extract below) 
Then first build and install OCPN as described in the same file.
(To make the build faster I though use to: cmake -DOCPN_BUILD_TEST:BOOL=OFF .. )
After that I can build plugins as well.
For the Debug part in Linux a can though not be on very much help.

From INSTALL:
On Debian/Ubuntu the canonical way to install build depencies is:
Pls CD to ...(Builds?)..\OpenCPN\
    $ sudo apt install devscripts equivs
    $ sudo mk-build-deps -i -r ci/control
    $ sudo apt-get --allow-unauthenticated install -f

rgleason, 2023-01-14:
Thanks Hakan!

rgleason, 2023-01-15:
Thanks all. Time to close this I think.

closed 2023-01-15 by rgleason