OpenCPN/OpenCPN #2930: Delete an active track and Freeze OpenCPN

rgleason, 2023-01-14:
**Describe the bug**
Delete and actively being made Track and freeze OpenCPN

Win 11 OpenCPN 5.7.1 in VS3022 debug mode.

rgleason, 2023-01-14:
CanvasMenu.cpp
Line 1769      RoutemanGui(*g_pRouteMan).DeleteTrack(m_pSelectedTrack);

Exception thrown at 0x76F3DCCA (ntdll.dll) in opencpn.exe: 0xC0000005: Access violation reading location 0x00000020.

The thread 0x58ac has exited with code 0 (0x0).
'opencpn.exe' (Win32): Loaded 'C:\Windows\WinSxS\x86_microsoft.windows.gdiplus_6595b64144ccf1df_1.1.22621.819_none_21dd7a56301966e7\GdiPlus.dll'. 
'opencpn.exe' (Win32): Loaded 'C:\Windows\SysWOW64\edputil.dll'. 
The thread 0x170 has exited with code 0 (0x0).
The thread 0x5488 has exited with code 0 (0x0).
The thread 0x5cc0 has exited with code 0 (0x0).
The thread 0x2840 has exited with code 0 (0x0).
The thread 0x2104 has exited with code 0 (0x0).
Exception thrown at 0x76F3DCCA (ntdll.dll) in opencpn.exe: 0xC0000005: Access violation reading location 0x00000020.

[image]

rgleason, 2023-01-15:
I also got another immediate break
>	[Inline Frame] opencpn.exe!pugi::impl::`anonymous-namespace'::xml_allocator::allocate_memory(unsigned int) Line 526	C++

```
The thread 0xb68 has exited with code 0 (0x0).
The thread 0x6394 has exited with code 0 (0x0).
The thread 0x6360 has exited with code 0 (0x0).
Exception thrown at 0x75857452 (KernelBase.dll) in opencpn.exe: 0x000006BA: The RPC server is unavailable.
onecore\vm\dv\storage\plan9\rdr\dll\util.cpp(99)\p9np.dll!5D7CEBC3: (caller: 5D7C81B9) LogHr(5) tid(40f8) C0000034     Msg:[瑎牃慥整楆敬☨敤楶散‬奓䍎剈乏婉ⱅ☠瑡牴扩瑵獥‬椦卯慴畴ⱳ渠汵灬牴‬䥆䕌䅟呔䥒啂䕔也剏䅍ⱌ⠠䥆䕌卟䅈䕒剟䅅⁄⁼䥆䕌卟䅈䕒坟䥒䕔簠䘠䱉彅䡓剁彅䕄䕌䕔Ⱙ䘠䱉彅偏久‬䥆䕌卟乙䡃佒低单䥟彏低䅎䕌呒‬畮汬瑰Ⱳ〠)] 
The thread 0x40f8 has exited with code 0 (0x0).
Exception thrown at 0x75857452 (KernelBase.dll) in opencpn.exe: 0x000006BA: The RPC server is unavailable.
onecore\vm\dv\storage\plan9\rdr\dll\util.cpp(99)\p9np.dll!5D7CEBC3: (caller: 5D7C81B9) LogHr(6) tid(57ec) C0000034     Msg:[瑎牃慥整楆敬☨敤楶散‬奓䍎剈乏婉ⱅ☠瑡牴扩瑵獥‬椦卯慴畴ⱳ渠汵灬牴‬䥆䕌䅟呔䥒啂䕔也剏䅍ⱌ⠠䥆䕌卟䅈䕒剟䅅⁄⁼䥆䕌卟䅈䕒坟䥒䕔簠䘠䱉彅䡓剁彅䕄䕌䕔Ⱙ䘠䱉彅偏久‬䥆䕌卟乙䡃佒低单䥟彏低䅎䕌呒‬畮汬瑰Ⱳ〠)] 
The thread 0x57ec has exited with code 0 (0x0).
The thread 0x4b88 has exited with code 0 (0x0).
Exception thrown: read access violation.
**alloc** was 0x12.

```

[image]

Hakansv, 2023-01-15:
Rick..
Just a Q: Is the same procedure creating the Debug exception working in runtime mode? --Can you delete an active track in runtime mode.
(In Win10 I can't reproduce this)

rgleason, 2023-01-15:
Yes, this does not happen in normal OpenCPN, sorry I should have tested that.  Also the problem may be associated with the plugin that I was testing  "Autopilot_route"  which appears to have a lot of problems, and will not be added to the master catalog.

Closing.

closed 2023-01-15 by rgleason