OpenCPN/OpenCPN #2931: Are there larger problems with PIM plugins?

rgleason, 2023-01-16:
See this report by Marco
https://github.com/marcobergman/ais_simulation/issues/10#issuecomment-1383234142

On Ubuntu, Update Catalog only shows me the plugins in the local directories; it does not seem to pull anything from the web.

On raspbian, I see this when I update Catalog:

20:21:54.277 MESSAGE OCPNPlatform.cpp:1101 ERROR Loading translation catalog for:  opencpn-pypilot_pi
ccc

rgleason, 2023-01-16:
If someone using these OS (Ubuntu and RPI can try installing the  pypilot_pi plugin to provide some insight as to what is wrong, it would be appreciated.

bdbcat, 2023-01-16:
Rick...
Just tested.  Seems that the beta catalog is empty for ubuntu 22.04. x86
I suggest you check the metadata XML files.

rgleason, 2023-01-16:
Thanks Dave.

What I've found is this
https://cloudsmith.io/~opencpn/repos/pypilot-beta/packages/?q=0.36.3.0+ubuntu&page=2   Cloudsmith package exist for pypilot ubuntu in opencpn/pypilot-beta
[image]

The metadata I have in my local plugins/beta branch is this:
[image]

And there are about 19 metadata for pypilot 0.36.3.0 in the OpenCPN/plugins  Beta Catalog
https://github.com/OpenCPN/plugins/tree/Beta/metadata

I am not sure what is happening.

When I paste

`https://dl.cloudsmith.io/public/opencpn/pypilot-beta/raw/names/pypilot_pi-0.36.3.0-ubuntu-x86_64-22.04-jammy-tarball/versions/0.36.3.0+1677.6ab4cd8/pypilot_pi-0.36.3.0-ubuntu-x86_64-22.04-jammy.tar.gz`

Into my browser it downloads.

rgleason, 2023-01-16:
Why is marco bergman not seeing the catalog?
https://github.com/marcobergman/ais_simulation/issues/10#issuecomment-1383234142

```
On Ubuntu, Update Catalog only shows me the plugins in the local directories; it does not seem to pull anything from the web.

On raspbian, I see this when I update Catalog:

20:21:54.277 MESSAGE OCPNPlatform.cpp:1101 ERROR Loading translation catalog for:  opencpn-pypilot_pi
ccc

```

rgleason, 2023-01-16:
I wonder if he is using the "master" catalog rather than "Beta"?

leamas, 2023-01-16:
The plugins have `<target>` set to *ubuntu-x86_64*.  This is strange,  the general contract these days is that plugins advertises themselves as something like *debian-wx32-x86_64* or so.

leamas, 2023-01-16:
@rgleason : Is pypilot based on shipdriver or testplugin?

rgleason, 2023-01-16:
Hi Alec, Pypilot is based on TP FE2.

Also I just checked for autopilot_route_pi 0.4.8.0 which Marco is also using, and it does exist in the  opencpn/plugins beta branch
https://github.com/OpenCPN/plugins/tree/Beta/metadata

and I also find that a search on  
https://cloudsmith.io/~opencpn/repos/autopilot-route-prod/packages/?q=0.4.8.0+ubuntu  yields about two pages.

rgleason, 2023-01-16:
Alec,
Anyway, the ABIs (that is, targets) we use from 5.8.0 are listed in the top comment of https://github.com/OpenCPN/OpenCPN/discussions/2797. ubuntu-x86_64 is not listed there, so this is a problem.

Ok, so what should I do or is this something he should do?
Does he have to install flatpak or something, because we don't support installation on ubuntu as an application?

rgleason, 2023-01-16:
Thanks Alec,
 
**I guess we need to change this, but this change affects every single plugin and I do not intend to rebuild them all, it is a pain in the butt and takes two days.**

Isn't there an easier way? This target has always been used I think.
 
> The plugins have <target> set to ubuntu-x86_64. This is strange, the general contract these days is that plugins advertises themselves as something like debian-wx32-x86_64 or so.

leamas, 2023-01-16:
If I get this right, this is either about you or the testplugin template,  I have no insight in the boundaries.

You need to make the catalog entry correct. I **think** that `<target>` should be *debian-wx32-x86_64* but you need to double-check this.  I think shipdriver has this  right, so you can check what shipdriver uses in the catalog.

Once you understand what it should be, you need to get a correct catalog entry. You also need to make sure the plugin can be imported. This is perhaps an issue  to file against testplugin, perhaps not. But in the end, `<target>` should be correct, in the catalog and when importing.

leamas, 2023-01-17:
> This target has always been used I think. Just make a ubuntu-x86_64 target.

No, we are now completing https://github.com/OpenCPN/OpenCPN/discussions/2502, which is about using debian-xxx instead of ubuntu-xxx targets. 

And *Just make a ubuntu-x86_64 target*... well, this is not that simple.  The code in OpenCPN which  checks the ABI is sort of locked now, we are after all in beta phase.  So...

rgleason, 2023-01-17:
Alec, I am not going to get into this discussion with you. Thank you for responding and clarifying the problem. 

There are now about 20 plugins that will not be available for quite awhile.  I am sorry about that but I am not going to redo them all.

leamas, 2023-01-17:
fair enough, it is about your time, nobody else makes decisions about that.

rgleason, 2023-01-17:
Yes, Thanks Alec, I do value my time.  I guess I was right about the title of this thread.

rgleason, 2023-01-17:
These plugins were pushed to the beta catalog, about 25 plugins. There was not a single report of a problem.  This process is totally busted.  Is anyone interested in using the plugins? Apparently not.  So why even spend the time?

rgleason, 2023-01-17:
This is a complicated process, and it is inevitable that there will be some problems. For TestPlugin FE2 templated plugins I would like to [implement the process listed here](https://github.com/jongough/testplugin_pi/issues/283).

Comments would be appreciated.  Is this OK Dave?

rgleason, 2023-01-17:
Now I get it, this is a work in progress still see
https://github.com/OpenCPN/OpenCPN/discussions/2502#discussioncomment-1923145

**Table of Targets & Template Build Information**

I have a request. When all of the "work" has settled down and appears to be somewhat static, can we have a complete update of this [Table of 5.8.0 beta dependencies.](https://github.com/OpenCPN/OpenCPN/discussions/2797) with all pertinent information about various OS required by a template to make "legal" plugins for that OS?

leamas, 2023-01-17:
What's missing? On phone.

bdbcat, 2023-01-18:
Rick...
The process you outline will be effective, even if a little ambitious. 
Organizing devs is one thing.  But organizing an ad-hoc group of casual testers is quite another. 
Also, I prefer motivating by carrot rather than stick, so #5 seems a bit, well, harsh.  I don't think you really mean that....
But feel free to move ahead with this plan as you wish, subject to real world factors as they arise.

rgleason, 2023-01-29:
This is in process now, can it be closed ?

rgleason, 2023-01-31:
Users are testing TP plugin for all the various OS. When that is complete we wil have a good template.  I think this can be closed. now.

closed 2023-01-31 by rgleason