OpenCPN/OpenCPN #2942: cannot cmake demo_pi

marcobergman, 2023-01-24:
Newbie question; don't know where to ask:

How can I compile demo_pi? I'm getting this error:

```
marcb@ubuntu-desktop:~/OpenCPN/plugins/demo_pi_sample/build$ cmake ..

[...]

CMake Error at cmake/PluginInstall.cmake:13 (target_link_libraries):
  The keyword signature for target_link_libraries has already been used with
  the target "demo_pi".  All uses of target_link_libraries with a target must
  be either all-keyword or all-plain.
```

Thx!

rgleason, 2023-01-24:
@Rasbats   ?? Can you help Marco?

Rasbats, 2023-01-25:
Never tried to work with demo_pi. Actually a basic working demo would be very useful. DR_pi has been used as a start point for another plugin:
[https://github.com/Rasbats/DR_pi/releases/tag/v4.0.0](https://github.com/Rasbats/DR_pi/releases/tag/v4.0.0)
@marcobergman : PM me if you need any help with compiling this one.

marcobergman, 2023-02-02:
Hey Rasbats, belated thanks for your reply. I don't want to misuse your
kindness or waste your time, but if you are one of those who know how to
create ocpn plugins you might be able to tell me where would be the proper
place to get support on plugin development.

I've cloned a plugin, searched-and-replaced to meet my ends, but now I get
this

[image: image.png]

and this:

 ERROR dlunix.cpp:162 /usr/local/lib/opencpn/libvhf_info_pi.so: undefined
symbol: _Z19PlugInHandleVhfInfob

I think these are related. Where could I get help?

Thx,

/Marco Bergman

marcobergman, 2023-02-02:
Hey Rasbats, belated thanks for your reply. I don't want to misuse your kindness or waste your time, but if you are one of those who know how to create ocpn plugins you might be able to tell me where would be the proper place to get support on plugin development.

I've cloned a plugin, searched-and-replaced to meet my ends, but now I get this

[image]

and this:

 ```
ERROR dlunix.cpp:162 /usr/local/lib/opencpn/libvhf_info_pi.so: undefined symbol: _Z19PlugInHandleVhfInfob
```

I think these are related. Where could I get help?

Thx,

/Marco Bergman

nohal, 2023-02-02:
This looks like an undefined function in your own code. Is it somewhere to be seen?
Either https://opencpn.zulipchat.com or https://github.com/OpenCPN/OpenCPN/discussions would be better places for continuing this kind of discussion, but still we will probably need to have a look at what you have done so far.

leamas, 2023-02-02:
@nohal :  

> This looks like an undefined function in your own code. Is it somewhere to be seen?

No, this is cmake used in a incompatible way. My first guess is that demo_pi has bit rot after sweeping changes to the
overall cmake build system.

One question is of we should maintain this, at least in the main project. Last time it got some love beyond formatting etc. seems to be around 2020. 

I'd suggest that we convert this to a Discussion item (link center-right). This would get it out of the way for the 5.8 release while still having a room to discuss what to do with demo_pi.