OpenCPN/OpenCPN #2943: DPI Aware OpenCPN  5.7.1+98.7941c00

rgleason, 2023-01-27:
Dave wrote:
> The general idea of Windows scaled display update is to enable most
> users to have a reasonable display without need to fiddle with font
> sizes.  The code will attempt to arrange things automatically, according
> to MS recommended best practices.
> 
> Additionally, I have added a slider to control ENC test feature size,
> for personal preference. It is still possible to change font sizes,
> faces, style, color, etc. manually.
> 
> With these changes, I think we can now declare that OCPN is "DPI aware",
> in MS terminology.
> 

**RESULTS**
[image]

[image]

[image]

[image]

bdbcat, 2023-01-27:
Clearly some trouble with tool sizes.  Investigating now.

bdbcat, 2023-01-28:
Rick...
I would like to see your logfile, please

rgleason, 2023-01-28:
Logfile sent via email about 1/2 hr ago also here. 
[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/10527323/opencpn.log)
[opencpn.ini.txt](https://github.com/OpenCPN/OpenCPN/files/10527327/opencpn.ini.txt)

Here are some screen shots with notes
[image]

[image]
[image]
[image]
[image]

rgleason, 2023-01-28:
Noticed one other.

[image]

rgleason, 2023-01-28:
Solution to extra-large toolbars:  Turn OFF settings->UI->"Enable Scaled
Graphics Interface"
 
Good point Dave.  User just needs to close OpenCPN and reopen again to get the full readjustment as the icons become very very small if you don't

[image]

closed 2023-01-29 by rgleason

rgleason, 2023-01-29:
https://github.com/OpenCPN/OpenCPN/pull/2940#event-8382045254