OpenCPN/OpenCPN #2946: Extremely slow panning / zooming performance on Win11

adam-ah, 2023-01-28:
**Describe the bug**

Some map areas on lower zoom level (1:311000 and closer) show extremely slow / unusable panning and zooming performance on OpenCPN 5.6.2.
The CPU is idling, OpenCPN uses at most 8% when panning. OpenGL enabled, high-end GPU GPU idling, no load.

**To Reproduce**
Steps to reproduce the behavior:
1. No plugins are enabled
2. Go to La Paz area on CM93
3. Zoom in

**Expected behavior**
Fluid panning and zooming expected on modern machines.

**Screenshots**

https://user-images.githubusercontent.com/3745745/215294622-8d8895ac-b0f3-419d-85c7-1331de53a373.mp4

**Desktop (please complete the following information if applicable):**
 - Windows 11

nohal, 2023-01-29:
Not reproducible for me on neither a "modern" nor "not modern" Windows machine. Please make sure your drivers are up to date and/or provide better HW information.

adam-ah, 2023-01-29:
>  drivers are up to date

yes

> provide better HW information.

CPU: 13600K
GPU: 3080
Mem: 64Gb
Free disk space: >1TB

nohal, 2023-01-29:
are you able to reproduce the behavior using the integrated Intel GPU?

adam-ah, 2023-01-29:
I have the _F_ version, without integrated graphics.
The problem is the same even if I turn off the "Use accelerated graphics (OpenGL)" option in OpenCPN.

nohal, 2023-01-31:
Not reproducible with current code (It is reproducible with 5.6.2). Please reopen if you are able to reliably reproduce with the current  development versions.

closed 2023-01-31 by nohal