OpenCPN/OpenCPN #2947: No release for almost a year ?

adam-ah, 2023-01-28:
**Describe the bug**
Last release was 280 days ago, 5.6.2.

Since then, there was 1598 commits.

Why aren't these changes being released?

bdbcat, 2023-01-29:
The changed _are_ all released, continuously, right here on github.

adam-ah, 2023-01-29:
They don't seem to be, latest release is 2022 April

[image]

nohal, 2023-01-29:
A numbered formal release is published when ready, not after an arbitrary number of commits or period of time. Every single commit to this repository is built and published by the CI workflows as you can trivially confirm for yourself by looking at https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/.

closed 2023-01-29 by nohal

bdbcat, 2023-01-29:
Changes are released as source code.
Build whenever you like.  Official Releases happen "when they are done".

adam-ah, 2023-01-29:
According to the [official OpenCPN documentation](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation), the way to obtain the latest version is to get the release from GitHub, which has not been updated in almost a year. The unstable CI packages are not referenced in the official documentation.

> Changes are released as source code.

Maybe this is a comprehension issue. Please refer to the official [GitHub documentation what a _release_ is](https://docs.github.com/en/repositories/releasing-projects-on-github/about-releases)

rgleason, 2023-01-29:
This is an open source project which depends on volunteers. You are invited to help update the documentation.

adam-ah, 2023-01-29:
Thank you for the offer, @rgleason, I initially wanted to after started using both the desktop and the Android version. However, after realising that OpenCPN repo owners do not release new versions, nor planning on releasing new versions on any of the supported platforms, I changed my mind and will not be contributing to the project.

nohal, 2023-01-29:
> nor planning on releasing new versions on any of the supported platforms

What exactly leads you to this conclusion please? Have you really completely missed all the alpha and beta versions for the upcoming 5.8?

rgleason, 2023-01-29:
Dear Adam, I think you are shooting yourself in the foot, and making a scene of it, but that's your choice. It is a good community, If you happen to change your mind, we would welcome some help.