OpenCPN/OpenCPN #2949: UI: Chart Object-bouy-route-soundings ENC Improvement

rgleason, 2023-01-29:
For ENC Charts
Chart Object Scale Slider controls:

- Chart Objects
- Bouys
- Soundings

Yet we have a nice ENC Sounding slider now.
Perhaps it should be used exclusively for Soundings?

[image]
[image]

bdbcat, 2023-01-30:
Corrected in github master.
Please build, test and close if possible.
Thanks

rgleason, 2023-01-30:
Works very nicely with ENC and CMP, did not try any other chart types, but I expect RNC is different.  Thank you.