OpenCPN/OpenCPN #2950: UI: Create Route rubberband and yellow area

rgleason, 2023-01-29:
In high res display Win11
There seems to be a missing rubber band when creating a route.
Should it be made thicker because it is too thin to show visibly on the High Res Screen?
[image]

rgleason, 2023-01-29:
[image]
[image]

bdbcat, 2023-01-29:
Cannot reproduce, works fine for me.
Hubert?

bdbcat, 2023-01-29:
Works as designed, close now, please.

closed 2023-01-29 by rgleason