OpenCPN/OpenCPN #2952: UI: Right Screen Menu Overlaps  (small bug)

rgleason, 2023-01-29:
[image]

[image]

[image]

[image]

**Maybe it should be cut off for PI menu items that are too long?**

[image]

bdbcat, 2023-01-29:
Fixed, will be in next pretest#3

rgleason, 2023-01-30:
Thanks, This appears to be fixed on Windows, most recent master compiled. Closing

[image]
[image]

closed 2023-01-30 by rgleason