OpenCPN/OpenCPN #2954: Not allow re-transmit on same Serial port

BerndCirotzki, 2023-01-29:
Hello developers,

can you please, please change this :      
 in File muliplexer-cpp :
      ..... 	  
252   //  Allow re-transmit on same port (if type is SERIAL),
      //  or any any other NMEA0183 port supporting output
      //  But, do not echo to the source network interface.  This will likely recurse...
	  
        if (params.Type == SERIAL || driver->iface != source_iface) {   ... change to    
		
        if (driver->iface != source_iface) {
      .....

Delete "params.Type == SERIAL" ... it makes so sense but a lot of traffic an Serial Output port.
If anyone want's to use it, it would be better to have a checkbox in Optionmenue->Connections Settings to activate it. (Idea ?)

thanks

Hakansv, 2023-01-29:
I may misunderstand but there is a checkbox for that? Default not enabled.

[image]

BerndCirotzki, 2023-01-29:
Oh yes, you misunderstand. This checkbox is only to define this port as Output port, an not to send the same Data back to his own Port.
The Checkbox "Receive Input on this Port" says, that the port in not only an output port.

So I get a Message as Input Message an send it direct back to myself. That makes no sense. But it makes sense, to Send out other sentences from other port,s or sentences from Plugins created ..

rgleason, 2023-01-29:
Just so we understand.  Bernd is this correct?

1. Checking the Output Port should not include sending data received back out (which is received again on the input port)?

2. Checking the Output Port should send sentences from other ports or sentences from Plugins....

Hakansv, 2023-01-29:
- If output is enabled on a port every NMEA0183 messages available by the multiplexer is sent out. Independent of the data source. That implies it is possible to output on e.g. a COM port what's received from the same COM port. Correct.
You may need to use the filter function to control desired messages for output. If you for example need only messages for an Autopilot set the filter to transmit only message where the talker is CD. See pict.
- In my own case onboard it does make sense to transmit what's received on the same port. The RX pin receive wind data. The TX pin is connected to the Autopilot where I need wind data. So I use the filter function to transmit $CDxxx and $xxMWV

[image]

BerndCirotzki, 2023-01-29:
Yes ... that's it  ... two Checkboxes ...
I think I would name them a little different ...

1. Checking the Output Port should not include sending data received back out (which is received on this input port)?
   ( and when it is NOT checked, the senteces will be mirred.)
2. Checking the Output Port should send sentences from other ports or sentences from Plugins....

....
If output is enabled on a port every NMEA0183 messages available by the multiplexer is sent out. Independent of the data source. That implies it is possible to output on e.g. a COM port what's received from the same COM port. Correct.
You may need to use the filter function to control desired messages for output. If you for example need only messages for an Autopilot set the filter to transmit only message where the talker is CD. See pict.
....
That's the way I cast around at the moment. 
But, waht do you do, when the GPS Antenna is on the same Port like the Autopilot ? 

In my design a Data come form ONE Port.   AIS, GPS, Instrument data ... an Autopilot. an everythis mus be filtered komplex.

BerndCirotzki, 2023-01-29:
.....
In my own case onboard it does make sense to transmit what's received on the same port. The RX pin receive wind data. The TX pin is connected to the Autopilot where I need wind data. So I use the filter function to transmit $CDxxx and $xxMWV
...

And you See, you use it others. And with two checkboxes, both would be helped.

Hakansv, 2023-01-29:
> But, waht do you do, when the GPS Antenna is on the same Port like the Autopilot ?

Set the output filter to Transmit sentence: "CD" and only the three CD-messages from a OCPN active route will be sent out. Nothing else.

BerndCirotzki, 2023-01-29:
... Ok, I thought it could be a feature.

BerndCirotzki, 2023-01-29:
In my own case onboard it does make sense to transmit what's received on the same port. The RX pin receive wind data. The TX pin is connected to the Autopilot where I need wind data. So I use the filter function to transmit $CDxxx and $xxMWV

... but when's it is the same Port, the autopilot has already the Wind data ? and when not, why

Hakansv, 2023-01-29:
> ... but when's it is the same Port, the autopilot has already the Wind data ? and when not, why

On the PC it's the same COM port connected by a serial to USB adapter. The serial RX wire is connected to a wind data sensor and the TX wire to the AP.  (Saving COM ports and adapters : ) )

BerndCirotzki, 2023-01-29:
Ok,
I see, differend designs. wouldn't it be nice to manage both ?!
with a second checkbox like discribed it would be able.

rgleason, 2023-01-29:
Would this primarily be used for all autopilots?

1. Checking the Output Port should not include sending data received back out (which is received on this input port)?
( and when it is NOT checked, the senteces will be mirred.

BerndCirotzki, 2023-01-29:
Would this primarily be used for all autopilots?  .....

I don't know, but why only autopilot ? 
I also send out RMC, GLL Messages to my Instrument, to know GOG, SOG .... and the Variation from WMM Pluging, send out from My "Autopilot Plugin" that could modfy the RMC witch is not an autopilot feature.

bdbcat, 2023-01-29:
It is a general case, when serial ports or USB ports are limited, and wiring is difficult.  Allows one serial port to support two "channels", one for input, and another for output.  Often used for autopilot, but not exclusively.
I am a little confused by this thread.
Please re-state the requested change.

BerndCirotzki, 2023-01-29:
Ok,
because there a several designs for Using the Serial conection, here is the new requested change like rgleason undersod:

Expand the "One" checkbox "Output on this Port ( as autopilot or NMEA Reapaeter)" to two checkboxes.
1.    Send (Mirror) incomming Data back to this Port as Output.(which is received on this input port).
      and second 
2.   Send Data to this Port as Output Port from other ports, internal sentences from Plugins or OpenCPN myself.

tvr256, 2023-01-29:
I personally think talking about checkboxes is confusing the matter.

The real change request is : If an NMEA sentence is received on a port, it should *not* be immediately retransmitted on the same port. This change can be implemented without any changes to checkbox behaviour.

I can't think of any use cases where retransmitting an NMEA sentence on the same port is useful, and in most cases it just causes loops.

BerndCirotzki, 2023-01-29:
Hello tvr256,
this I thougth myself too, but see how "Hakansv" uses this ! ... serveral designs.
Now, I know why it is so since a long time. And with a new checkbox both designs can be used.

tvr256, 2023-01-29:
> Hello tvr256, this I thougth myself too, but see how "Hakansv" uses this ! ... serveral designs. Now, I know why it is so since a long time. And with a new checkbox both designs can be used.

You (and HakanSV) are absolutely right, I can see now why sometimes you might want to retransmit on the same port. So yes, a checkbox would be needed.

Hakansv, 2023-01-29:
I myself can't see a reason for another checkbox.
Nothing is sent out unless "Output to this port.." is enabled.
If you want to send out anything you in most cases need to limit what's sent with a filter. 
Unless you make a UDP broadcast you for sure doesn't want to send out all including e.g. AIS targets, so you have to use the filter.

BerndCirotzki, 2023-01-29:
I myself can't see a reason for another checkbox.
... For you it is OK that is what I undersand. 👎
Ok. My Autopilot Plugin receives "$STALK" Sentences an receives "$STALK" sentences ... witch one will you filter ?

BerndCirotzki, 2023-01-29:
Ok. My Autopilot Plugin receives "$STALK" Sentences an sends "$STALK" sentences ... witch one will you filter ?

Hakansv, 2023-01-29:
"ALK" never heard of but if received from another source and the AP needs it and you don't want the mystery return, filter it out on the input.

BerndCirotzki, 2023-01-29:
translated Seatalk Sentence.

https://www.boatoon.com/de/shop/seatalk-nmea-konverter-iso-22364.html  .... for Example.

I buld my own, that also translates to N2K.

BerndCirotzki, 2023-01-29:
Goolge search "STALK NMEA"  ...
$STALK,84,26,A2,88,40,00,FE,02,06*15  ... Example for Autopilot Status. Needed as Input. could not filter
$STALK,86, ...... Send from Plugin as Keystroke.   could not filter !

bdbcat, 2023-01-29:
Why can you not filter $STALK?

BerndCirotzki, 2023-01-29:
Why is it not possible to add a checkbox?

bdbcat, 2023-01-30:
Because we normally do not change the long established GUI and program logic for one edge case arriving late in a Beta cycle.
If you must have this function, I suggest you fork and build your own custom version of OCPN.

BerndCirotzki, 2023-01-30:
Ok, then I build my own. This makes me a lot of work, every time to build my own OpenCPN.
PS : on NMEA2000 this will cause fatal errors.

So I tested your new N2k Connection. Ok this needs much work.