OpenCPN/OpenCPN #2957: UI: PI Dashboard Preferences

rgleason, 2023-01-29:
Win11 High Res Screen

Preferences Icons are out of scale and too big.
Also see other improvements to the menu.
Taken from a closed Issue.

[image]

rgleason, 2023-02-11:
This needs some attention

rgleason, 2023-02-19:
https://github.com/OpenCPN/OpenCPN/pull/3012

Hakansv, 2023-02-20:
This and #2983 would be adjusted at latest master. Pls test.
The spaces commented above is not really waste of estate. The size of this page is controlled by the "Appearance" page needing this size to view all options.

rgleason, 2023-02-20:
I don't understand the comment "The spaces commented above is not really waste of estate. The size of this page is controlled by the "Appearance" page needing this size to view all options."

It is still much the same.
[image]

Hakansv, 2023-02-20:
> It is still much the same.

Strange.. If you build locally, when did you pull? The patch for this was merged tonight(UTC). 

Space.. The overall size of "Dashboard preferences" is controlled by what's in tab "Appearance". Tab "Dashboard" use that space and expand what's needed.

rgleason, 2023-02-20:
Some comments made before your post, below. I will try to pull again.
[image]
[image]
[image]

Hakansv, 2023-02-20:
Rick..
One thought based on own experiences:
If you build in VS2022 and make a installer package remember to always use "Rebuild only package". If you just "Build" the changes in plugins, e.g. Dashboard" may not be included in the package.
[image]

rgleason, 2023-02-20:
Thanks.  I had to highlight "Dashboard" in File Explorer and right click  Rebuild.
For some reason I don't have "PACKAGES" selections in VS2022

This is much much better. Thank you.  Perhaps some reduction in widths?

[image]

closed 2023-02-25 by rgleason