OpenCPN/OpenCPN #2959: NMEA2000 : OpenCPN is not sending Out (mirroring) N2K messages

BerndCirotzki, 2023-01-30:
when Opencpn is receiving N2K Data the messing will not be send Out to N2k Output port.
for example when controling an autopilot on N2k you have to send special PGN's.

bdbcat, 2023-01-30:
OpenCPN core does not transmit on N2K for any basic functions. 
OpenCPN does not echo nor multiplex N2K traffic.
OpenCPN core does not translate N2K to NMEA0183.  This can be done by plugin.
N2K Autopilot control will be by plugin, since there are no generally accepted A/P control PGNs.  All are vendor specific.
All this is by design.

BerndCirotzki, 2023-01-30:
Ok, n2k is jot ready yet.

closed 2023-01-30 by BerndCirotzki

rgleason, 2023-01-30:
I have started a page in the manual, here for anyone who knows more, to add details about the implementation.

https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:advanced_features:nema2k

leamas, 2023-01-30:
Implementation details are not end user info, they belong in the Developer Manual.

Here, we need to integrate MESSAGING.adoc as a chapter in Plugin Development. We also need an overall description including the sketches from the initial PR in OpenCPN Development.

rgleason, 2023-01-31:
Actually, Alec there needs to be User information too, otherwise you will have all kinds of questions and misunderstandings as witness in Issues here. 
I have distantly followed the process and development but am not qualified to write anything about this for Users. To bury this in some development doc does not make sense to me.