OpenCPN/OpenCPN #2960: MEA2000 : pointless filter mechanism connections settings. No PGN filter

BerndCirotzki, 2023-01-30:
when Protocol NMEA2000 is selected the filter is able to filter NMEA0183 sentence Names.
But ....
there are no sentence names in N2k ... N2k uses PGN.
So you could not filter N2K.

leamas, 2023-01-30:
N2k messages are only available using the new API.s available in 1.18. These are based on that receiver subscribes to messages rather than that everything is sent and filtered as required. Using the "old" APi which was used for NMEA0183 is just not supported.

There is provisionary documentation on the new API in the file MESSAGES.adoc.

On a sidenote, under the hood all messaging is done using these new routines. The "old" API is re-implemented on top of the new stuff. That is,  it is an alternative to use the new routines instead of the old stuff also for N0183.

rgleason, 2023-01-30:
Yes, as I understand it, OpenCPN, just handles the PGN essential to itself, and there is no debug window for N2K.  Plugins are free to implement an n2k interface.
Ah, Leamas beat me to it. His is more precise.

bdbcat, 2023-01-30:
No need to wait..
OpenCPN 5.7.1 listens to and processes all common PGNs now.  This is useful to many current users now.
Which is to say, OCPN d5.7.1 does what it was designed to do, absent specific bugs.

BerndCirotzki, 2023-01-30:
Ok, when its enougth for you.

closed 2023-01-30 by BerndCirotzki