OpenCPN/OpenCPN #2962: NMEA 2000 : No N2k Protocol on Network connection.

BerndCirotzki, 2023-01-30:
No N2k Protocol on Network connection.

bdbcat, 2023-01-30:
There is no generally recognized industry standard N2K protocol on TCP or UDP.  There are some vendor specific implementations.  These are not widely supported nor documented.
Perhaps later....

leamas, 2023-01-30:
> There is no generally recognized industry standard N2K protocol on TCP or UDP. 

The closest to that would be NMEA2000 which is based on ipv6.  This is somehow on the roadmap, for a Later Version ™

nohal, 2023-01-30:
I suppose you mean NMEA OneNet. There will be quite some fun with it as this time it actually does implement security and encryption, so reverse engineering it will be much more of a problem than it was for 0183 and 2000...

closed 2023-01-30 by BerndCirotzki

leamas, 2023-01-30:
> I suppose you mean NMEA OneNet

yes