OpenCPN/OpenCPN #2963: API AddSingleWaypointEx(p_waypoint) does not display range rings

antipole2, 2023-01-30:
**Describe the bug**
If a waypoint is added through the AddSingleWaypointEx(p_waypoint) API, any range rings are added but not displayed.
If range rings are added to a waypoint previously without any using the UpdateSingleWaypointEx(p_waypoint) API, they are added but not displayed.

If you open the Attributes panel, you can see that the rings have been created.  When you OK to dismiss the panel, the rings are displayed, even if no changes have been made.

If range rings are updated through the UpdateSingleWaypointEx(p_waypoint) API, they are immediately updated in the display.

Hypothesis: The rings are not being displayed when the first is added.

**To Reproduce**
The following script for the JavaScript plugin demonstrates
```
waypoint = {	
	markName: "Ring Test",
	isNameVisible: true,
	iconName: "anchor",
	nRanges: 2,
	rangeRingSpace: 0.5,
	rangeRingColour: "#0000FF"
      }

OCPNonContextMenu(go1, "Add test waypoint");
alert("Use context menu to add test waypoint");

function go1(clickPosition){
	alert(false);
	waypoint.position = clickPosition;
	OCPNaddSingleWaypoint(waypoint);
	print("You need to open and OK the waypoint properties window to see the range rings\n");
	}
```

**Expected behaviour**
The range rings should be displayed when the waypoint is added or updated to include them.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: MacOS Venturer.  Also occurs on Windows
 - OCPN Version n5.6.2

antipole2, 2023-03-01:
I have tested this, both add and update and confirm they  now work correctly on OCPN5.7.1 beta 4
MacOS

bdbcat, 2023-03-01:
Close?

antipole2, 2023-03-02:
Yes - we can close this now.

closed 2023-03-02 by antipole2