OpenCPN/OpenCPN #2964: Scaled Win: Pls scale tide and current data

Hakansv, 2023-01-31:
**Describe the bug**
Tide and current values on screen are to small
**To Reproduce**
1. Enable Tide and/or Current
2. See the values beside the graph

**Expected behavior**
Font size for tide and current would be sized by the same logic(slider) as soundings.
**Screenshots**
See shot

**Desktop (please complete the following information if applicable):**
 - Win10-11

[image]

transmitterdan, 2023-01-31:
Should these also be "blanked" at higher scale factors the same as soundings do today?

bdbcat, 2023-01-31:
I think he means "Should we apply SCAMIN" to tide bar values?"  I think not.  SCAMIN is an ENC concept.  Tide bars are more general, usable on RNC as well.

Hakansv, 2023-01-31:
The slider use was just a thought to consider, as you did.
Thanks.

closed 2023-02-02 by Hakansv

rgleason, 2023-02-02:
Is this fixed?   Yesterday's master doesn't have it fixed.

Hakansv, 2023-02-02:
Commit: 33e167834  Did it for me.

[image]

rgleason, 2023-02-02:
Yes, found them. It is better.