OpenCPN/OpenCPN #2967: User Manual - Documentation needed

rgleason, 2023-02-01:
**Prioritization of data sources for OCPN core.**

See https://github.com/OpenCPN/OpenCPN/issues/2745#issuecomment-1410848161

Edit the current manual
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic
[GNSS Setup](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:gps)
Connections
Connections- Multiple GNSS Input Filtering

Add the Priority Table, screenshots and we need an explanation.

**Changes made to support signalK and nmea2000**

See https://github.com/OpenCPN/OpenCPN/issues/2745#issuecomment-1412001436

Users have shown that they need some support documentation about this subject matter because they make all kinds of incorrect assumptions about what has been done, and then think that OpenCPN is not working properly.

- guidance about what to expect
- how it can be used
- explanation for the user that plugins must be "enabled" and specifically coded to use these new sources of data 
- table or list of plugins that are so enabled with a link to those plugins documentation for how to use them with that data.

rgleason, 2023-03-30:
Hakan, is this done and should be closed?  I know you've been busy, and have made some changes in the wiki.

rgleason, 2023-04-28:
@Hakan  Are these changes complete in the wiki now?  So I can close this?

Hakansv, 2023-05-01:
@rgleason 
What I did was a small portion about priorities. The list of needed changes covers more than that. 
I'm though not sure I'm the one to make it understandable. 
But we may not close the issue and the edited milestone would indicate just that?

rgleason, 2023-05-01:
Thanks, Hakan. Best to keep it open I think.

rgleason, 2023-10-01:
Hakan, did this get completed?  or is there more to do?

rgleason, 2024-03-28:
GNSS Changes
I've added several paragraphs to the beginning. I would appreciate some review and any suggestions.
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:gps#manufacturer_gps_gnss_devices_and_software

rgleason, 2024-03-28:
Folks you've been programming your hearts out and I'm concerned that nobody is going to "get it".
In particular, "[Connections](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:connections#connections)"
Is the main page OK? Where does it need to change?
Are the sub pages acceptable and represent what is needed for each subject?
Is this going to be ready when the program goes out?  I think the documentation needs some help and I am not the right person to do this part.

While I am at it what should we do about these sections which have not been brought into the "Connections" page?
FIXME Not Transferred
[Connection Notes](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:options_setting:connections#connection_notes)
[Sending an Active Route to an Autopilot](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:options_setting:connections#sending_an_active_route_to_an_autopilot)
[Sending Routes and Waypoints to a GPS](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:options_setting:connections#sending_routes_and_waypoints_to_a_gps)
[Broadcast and Multicast](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:options_setting:connections#broadcast_and_multicast)

Should I proceed with that or not???
Thanks.

bdbcat, 2024-03-28:
Rick....
What is changed is the basic UI for adding, editing, and removing connections.
So I would start with **new pictures of the UI in action**.  You will find this implemented in github master.
When those are in place, we can look at details.
Probably most of the detail regarding connection parameters is unchanged .

The 4 other sections mentioned should surely be with the Connections section of the manual.
Move them verbatim, for now, following the basic connections UI documentation.

rgleason, 2024-03-28:
Thanks, 
The 4 sections are now sub pages, have added this to the [Contents Index ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:connections#contents) at the top of the page to bring them to readers view.

Will make new screenshots for Connections next.

rgleason, 2024-03-29:
Dave, I've updated the screenshots in this page https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:connections  I thought I had done this not so long ago, but now it should be ok.  

There are a couple of strange things about Connections.
[image]

Consistency between Input and Output interface.
[image]

rgleason, 2024-03-29:
This page appears to be wnanting some work.
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:connections:nmea2000#supported_nmea_2000_parameter_group_numbers_pgn

rgleason, 2024-04-02:
I've created a [User Manual Basic Discussion](https://github.com/OpenCPN/OpenCPN/discussions/3776) page  which is perhaps more appropriate.

rgleason, 2024-04-04:
Attention Please!
Does the Entire [Toolbar > Options > AIS page in the OLD User Manual ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toolbar_buttons:ais)qualify an [Advanced Topic to be located in User Manual Advance](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_advanced)  under Program?

rgleason, 2024-04-07:
Please review the main structure as we have it now. Is is OK?

[User Manual Basic](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic)
[User Manual Advanced](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_advanced)

The main question here is does the Structure (outline) of the pages serve our purposes?
Any comments would be appreciated.

I keep finding bad links, but now I think we have cleaned up 99.5% of them and these manuals are standing alone. I'd like to keep the OLD around for awhile until we finish editing.

One clarification.  We have ["Toolbar >Tracking"](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:tracking)
What is [Autofollow and Lookahead ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:chartbar:autofollow) then?  Are these older terms?
I also have a link under  [User Interface Basic > Autofollow](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:user_interface:auto_follow)

Hakansv, 2024-04-07:
I've been slightly checked the left table of contents. It looks really nice now. 
I've not yet examined all text but found so far some minor comments:

1. Toolbar description [here](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:user_interface#user_interface_basic) It describes how the main toolbar can be dragged. At least on my Win that's no longer possible. It stays all time at the leftmost chert edge.
2. GNSS setup. I think for most users a serial USB GNSS device is the first try. And also for most users is the simple method working. a) Connect the USB. b) Go to Options->Connections->Add new connection. c) Select Serial and use the drop down control and find the new USB. d) Select baude rate (very often 4800) OK - and your done. This would be uppermost and all details below are good but normally don't needed for the first time user. 

I'll check more of your above tasks later. You're making a huge job with this!

rgleason, 2024-04-07:
Thanks Hakan. Will address those.

Need to dispense with these then Toolbar > Options > Display is done[
Need to find a place for these](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:display#find_the_home_for_items_below
[Disable Full screen quilting](https://opencpn.org/wiki/dokuwiki/doku.php?)
id=opencpn:manual_basic:toolbar:options:display#disable_full_screen_quilting) - outdated no longer used?

rgleason, 2024-04-07:
@Hakansv 
Toolbar description [here](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:user_interface#user_interface_basic) It describes how the main toolbar can be dragged. At least on my Win that's no longer possible. It stays all time at the leftmost chert edge.

-- Win11    I am able to move it (slowly) with a left mouse button down, and it attaches to left and right sides of the Opencpn screen.

GNSS,  Thanks added this https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:gps#try_quick_start_method

Hakansv, 2024-04-07:
> One clarification. We have ["Toolbar >Tracking"](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:tracking)

No comments

> What is [Autofollow and Lookahead ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:chartbar:autofollow)then? Are these older terms?

I've made some edits. Pls check.

> I also have a link under [User Interface Basic > Autofollow](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:user_interface:auto_follow)

Works but maybe a bit surprising.  Could the contents "Auto Follow" link directly instead?

Hakansv, 2024-04-07:
> - Win11 I am able to move it (slowly) with a left mouse button down, and it attaches to left and right sides of the Opencpn screen.

I'm on present master code. I can't move the tool menu bar. Neither on Win nor Linux(Debian). Anyone else?
If it's as for me for others I don't judge this a bug. I see no need for that function?

bdbcat, 2024-04-07:
Correct.  Toolbar on O5.9+ is fixed in position, not possible to drag.  By design.

bdbcat, 2024-04-07:
Overzoom suppress by blur and suppress by scale are no longer used.

rgleason, 2024-04-08:
### Move Toolbar  - Remove it?
Why does my O5.9 work moving toolbars?  Do I need a new updated version?  Which one please?
Does moving the Toolbar work in any OS now?  Shouldn't we remove these notes then?
### Autofollow link in User Interface - Need a direct link (I haven't figured this technique out yet or if it is available)
link under [User Interface Basic > Autofollow](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:user_interface:auto_follow)
### [Autofollow and Lookahead ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:chartbar:autofollow)
Thanks looks much better.

Hakansv, 2024-04-08:
> Why does my O5.9 work moving toolbars? Do I need a new updated version? Which one please?

Take the most recent [here](https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/): 

> Does moving the Toolbar work in any OS now? Shouldn't we remove these notes then?

No -  Yes.

rgleason, 2024-04-09:
I'm trying to get user Editors set up with Opencpn 5.9 and I can't even tell which file to use, how will they know?
They have various equipment and OS:
- Win 10, 11
- Windows 10pro 
- Linux Min
- MAC LapTop Sonoma 14.3.1
- Android 8.1 on Galaxy Tab A   (secondary use)  - not available in version 5.9

Could you suggest which file they should download?
I'd like  to get this right the first time with them.
Thanks
-

bdbcat, 2024-04-09:
Rick...
Couple of things:
1.  At the moment, there is no "right the first time".  The source code is still changing, the final feature set is in flux, and for sure there are things still changing in the UI.
2. What we have on any given day is the latest build for Win10/11 and MacOS Sonoma.
3. We do not have a generic Linux Mint CI build.
4. We do not have any CI build for Android vxx at all.

What I would suggest:
1.  Teach the Windows and Mac editors to download and install the latest build, either daily, or every few days.
You know what that is for Windows.  For MacOS, it is the "universal" pkg, e.g 
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.9.0-universal+8719.3a57e6d9a-14.pkg#_gl=1*1h4aupx*_gcl_au*NTc1ODcyODM5LjE3MTI1Mzg5ODU.

2.  For the Linux Mint user:  If he is using O584 now, how did he install it?  We need to know that in order to point him toward the correct build.

All this uncertainty is quite normal at this point in Release cycle, for any big project.  Creating accurate documentation under these conditions is a challenge, I know.  But we can do it if we stay on track.

rgleason, 2024-04-09:
One Editor with Linux Mint found a deb file that works.  So he knows what works.
Thanks, I will advise them all about this using your writeup.

closed 2024-05-02 by rgleason