OpenCPN/OpenCPN #2970: Crash when zoomed out on RNC charts

transmitterdan, 2023-02-01:
**Describe the bug**
Windows latest master crashes when zooming out on RNC charts.

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest master branch build
2. I have RNC charts of the west coast of US latest versions
3. Set "course up mode" and "look ahead mode"
4. Have source of nav data active to drive the ownship position and course
5. Zoom out until you see the underlying world outline--O crashes

**Expected behavior**
O should not crash

**Screenshots**
N/A

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Version: 11

**Additional context**
This function
```
LLRegion ViewPort::GetLLRegion(const OCPNRegion &region)
```
appears to be the cause of the crash.  It trashes the stack so a traceback is difficult but I'm pretty sure this is the function active when the stack is corrupted.  It does it on all build types (Release/Debug/etc.).  I think it is related to the fact that the chart is "skewed" from north-up position.  If you try this with "North Up" mode it does not crash.  Only when the chart is tilted on the canvas does it crash.

bdbcat, 2023-02-02:
Could not reproduce immediately.  I guess we need more info.
Exact center location, charts visible.
Perhaps a VDR recording of NMEA data stream at time of crash, if repeatable.

transmitterdan, 2023-02-02:
@bdbcat I have it cornered here within the function:
```
        for (int k = 1; k < splits; k++) {
          float d = (float)k / splits;
          GetLLFromPix(wxPoint((1 - d) * p[li] + d * p[i],
                               (1 - d) * p[li + 1] + d * p[i + 1]),
                       pll + j, pll + j + 1);
          j += 2;
          wxASSERT(j < 539, _T("j is too large.")); // <----Added this wxASSERT which traps overflow of the pll array
        }
```
The variable j exceeds the bounds of pll which is a stack based double array of length 540.  This happens because the span of the viewport is really large (~130 degrees) and the 2-degree division of the latitude range ends up creating the variable 'splits' of 65.  Which means the pll array would need to be > 540.

I changed the size of pll to:
```
    double pll[640];
```
and the crash goes away.  But i'm not sure this is sufficiently large for all cases.

bdbcat, 2023-02-02:
Dan...
Great work.
I think it is logically impossible for splits to become larger than 180, so best to increase the array size to ((180 *  2)  + 2) * 8 = 2896.
It is no problem to have a (reasonably) extra large array on the stack. No performance hit at all, just BP pointer math.  Modern linux and Windows build systems typically allocate at least 1 MB on stack, and this function in not called recursively.  So safe enough.
I'll do that.
Thanks
Dave

closed 2023-02-02 by transmitterdan