OpenCPN/OpenCPN #2973: Grib  changes scale when turned on?

rgleason, 2023-02-02:
Is Grib_pi supposed to change scale when turned on from toolbar?

[image]
[image]

Ptizef, 2023-02-12:
Rick
Is it on Windows ?
I can't reproduce
JP

rgleason, 2023-02-13:
This is on Windows 11.
I just tried  opencpn_5.7.1+114.9715290_setup.exe win 11
and it still happens as reported.

rgleason, 2023-02-13:
Is this PR to fix the problem ?
https://github.com/OpenCPN/OpenCPN/pull/3005

rgleason, 2023-02-13:
Just pulled the opencpn master and built Opencpn.
This version does the same thing.  Load Grib_pi and you will be zoomed to somewhere else in the world.

Hakansv, 2023-02-13:
Rick..
No changes you've the "Zoom to file center...." option enabled and your grib file's center is "somewhere else in the..."?

[image]

Ptizef, 2023-02-13:
I use W11 and a compilation of the last master
I'm sorry, until now I can't reproduce the change of scale problem.
The PR 3005 is only an answer to issue 2773
JP

rgleason, 2023-02-13:
Hakan you are exactly right!  I feel like I am learning OpenCPN anew!

[image]

closed 2023-02-13 by rgleason