OpenCPN/OpenCPN #2974: Crash on exit

transmitterdan, 2023-02-02:
**Describe the bug**
Have seen a few crashes on exit.

**To Reproduce**
Steps to reproduce the behavior:
1. Not reproducible on demand

**Expected behavior**
No crash

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Version 11

** Backtrace **
```
>	opencpn.exe!S57Obj::~S57Obj() Line 101	C++
 	[External Code]	
 	opencpn.exe!s57chart::FreeObjectsAndRules() Line 486	C++
 	opencpn.exe!s57chart::~s57chart() Line 288	C++
 	[External Code]	
 	opencpn.exe!ChartDB::DeleteCacheEntry(CacheEntry * pce, bool bDelTexture, const wxString & msg) Line 261	C++
 	opencpn.exe!ChartDB::DeleteCacheEntry(int i, bool bDelTexture, const wxString & msg) Line 268	C++
 	opencpn.exe!ChartDB::PurgeCache() Line 277	C++
 	opencpn.exe!MyFrame::OnCloseWindow(wxCloseEvent & event) Line 1848	C++
```

nohal, 2023-03-30:
@transmitterdan is it still happening?

transmitterdan, 2023-03-30:
Yes, but not always with the same backtrack.

closed 2023-04-21 by transmitterdan