OpenCPN/OpenCPN #2988: Timeout? with large peer object transfers

nohal, 2023-02-04:
When trying to transfer a few big objects (1700 miles in 8 tracks in my case) from a "slow" machine, the client fails with error code -1 while the receiving server displays the PIN code, probably a timeout is reached somewhere.

bdbcat, 2023-02-09:
And where do you see the erro code (-1)?
Maybe here?
     err_msg.Printf("Server HTTP response is: %ld", response_code);
      OCPNMessageDialog mdlg(NULL, err_msg, wxString(_("OpenCPN Info")),
                          wxICON_ERROR | wxOK);
      mdlg.ShowModal();

Or?

nohal, 2023-02-09:
The problem boils down to this on the server side:

```
3dd2aa61 3 mongoose.c:4233:read_conn    2 0x15 snd 0/0 rcv 3139584/3141632 n=2048 err=0
3dd2aa76 3 mongoose.c:4233:read_conn    2 0x15 snd 0/0 rcv 3141632/3143680 n=2048 err=0
3dd2aa7c 3 mongoose.c:4233:read_conn    2 0x15 snd 0/0 rcv 3143680/3145728 n=2048 err=0
3dd2aa7d 1 mongoose.c:404:mg_error      2 0x15 max_recv_buf_size reached
3dd2aaea 3 mongoose.c:3416:mg_close_con 2 0x15 closed
```

nohal, 2023-02-09:
Seems like we need to process each `MG_EV_HTTP_CHUNK `, not just the final `MG_EV_HTTP_MSG`

nohal, 2023-02-09:
Sure yes, with tracks. I have some originally from Navionics that are 100MB+

bdbcat, 2023-02-10:
"process each MG_EV_HTTP_CHUNK "
Yes.  I'll dig into it this weekend.

bdbcat, 2023-02-15:
Pavel...
I implemented chunked transfer.  Not well tested on huge objects, but the logic flow looks right.
In master now.
A couple of things I would like to do with this, but have to be low priority now.
1.  Add a progress bar for the SendTo dialog on transmit progress, so user will not see dead screen on large transfers.  This will require a refactor, so that the dialog can get callbacks from curl by wx message loop.  Fiddly...
2. On large transfers, arrange to send a short "null data" transfer first, to establish the PIN if not yet known.  It would seem unfriendly to wait some minutes for the bulk transfer to complete, and then get the silly PIN dialog after all that.

Anyway, interested in your experiences with this patch on large data sets.

nohal, 2023-02-15:
Works quite fine. I would probably add

3. When the transfer finishes successfully, inform the user about it, just closing the dialog might lead to the user wondering what actually happened.
4. If the Route & Waypoint Manager dialog is opened on the server side during the transfer, it should be refreshed after the objects are received.
5. When happy with it, disable all the debug info going to `stdout`

Hakansv, 2023-02-19:
If the work on this is not finalized the canvas, right click, menu would benefit a "Send To" (peer) for tracks and WP as now for routes?

bdbcat, 2023-02-19:
Not finalized, should be easy to "clone" the route code.
Will do.

Hakansv, 2023-02-20:
Successfully tested for a track and WP on Win. Thanks.

bdbcat, 2023-02-25:
Close?

closed 2023-02-25 by nohal