OpenCPN/OpenCPN #2994: Creating route using near waypoint.

RooieDirk, 2023-02-09:
Current master on linux 22.04 GTK3.
If I want to make a new route using existing waypoints, then after clicking yes to "Do you want to use nearby waypoint", the chart keeps panning with the mousecursor movement. Also the start waypoint does stay underneath the mouse always and I can't move it to a destination.

Making a route with new waypoints goes fine.

bdbcat, 2023-02-09:
Corrected in github master now.
Thanks
Dave

RooieDirk, 2023-02-10:
Sorry Dave, Don't know what you did, but it did't help

bdbcat, 2023-02-10:
Do you by chance have Touch screen enabled in OCPN?

RooieDirk, 2023-02-10:
No, touchscreen is off. 
There is a small 'improvement' however as I can now draw a route with a max length of 90meter. Above 90m the chart starts panning again.
Did a touchscreen test and with touchscreen enabled it works as it should. (At least using a mouse as I don't have a touchscreen)

bdbcat, 2023-02-11:
Sorry, cannot reproduce anything like this.
Can you send some screenshots, or maybe a movie clip, to demonstrate the problem?

RooieDirk, 2023-02-11:
A short screencast.
I tried also change the "Lock waypoint" option, but no change.
https://user-images.githubusercontent.com/2668258/218247475-5d83cf80-5345-4959-95f5-5b7e30e24453.mp4

bdbcat, 2023-02-11:
Dirk...
I did finally reproduce this effect with a fresh virtual Jammy instance.
Corrected in github now, at least for me.  Take a look, if you please.

RooieDirk, 2023-02-11:
Much better. I can make a route now on my machine also. A minor issue remains, after clicking 'end route', the chart drag/pan starts again. A simple  mouseclick will stop it. (I can live with that;-))

bdbcat, 2023-02-11:
Dirk..
Very good. 
This bug is clearly related to wxWidgets 3.2 on GTK3.  Another instance of undocumented platform specific library variances.

RooieDirk, 2023-02-13:
Dave,
I'm completely happy now;-) Thanks

closed 2023-02-13 by RooieDirk