OpenCPN/OpenCPN #2997: Incompatible plugin message hangs OCPN

Hakansv, 2023-02-10:
**Describe the bug**
When opening Pluginmanager a message about "Incompatible plugin...." something showed up for some seconds. Then disappeared and Pluginmanager is hanged and so is OCPN.

**To Reproduce**
Steps to reproduce the behavior:
1.Open Option->Plugins
2. A message appears. What I could see before disappearance it was about some plugin and some path ... 
3. OCPN is badly hanged. Must kill it down.

**Expected behavior**
A clear and concise message that inform the user what to do.
Don't hang OCPN.

**Desktop (please complete the following information if applicable):**
 - OS: WIn10
 - Version Today's master

leamas, 2023-02-10:
I and Dave is having a private discussion on these issues ATM. The basic message is that the current state is kind of a transition while getting rid of old not well defined code.

I assume that you for now are able to work around these issues by removing plugins "by hand", right?

Hakansv, 2023-02-10:
Without having looked in the code I suppose it's I can work around. That's not the problem. 
I suppose it's about old plugins in "C:\ProgramData\opencpn\plugins"? If so something is not correct. There are no dll files present in my system. Only folders.

From the short glimpse I saw of the message I also suppose all user won't understand what it's all about. Most have learned to use the Pluginmanager when handling plugins. So he would open Options->Plugins and uninstall the plugin mentioned? And next time OCPN is opened the message will be there again.
The intention to get rid of old plugin binary files is of course good. We have seen some trouble created by these files. 
The first may be to ignore these files when detected so they don't disturb OCPN.
Next is to advice the user to delete them. Then we need to describe how to do that or maybe offer a function to delete the files in question? A installation program from OCPN have once installed them so would also be able to delete. If possible by a still present uninstall program.

leamas, 2023-02-10:
I'm not saying here is no bug, there certainly is  (and this is not the only one). Just hoping that you can handle the situation until I and Dave have sorted this out.

Hakansv, 2023-02-10:
I can possible handle it as long as I know what it's all about. The message(s) was not shown long enough to read them.
This code wouldn't have been tested enough before PR? 

To instead discuss the intention:
Looking in the log file the messages all are about plugins in ......\AppData\Local\opencpn\ that's maintained plugins and installed by Pluginmanager.  Neither of them are enabled so the message would have been shown only if I try to enable any of them. The presence of old plugins in that folder OCPN would be able to accept. Since it's not enabled it will not be started and also not disturb?
If I update the plugin catalog and install valid plugins all of these files would be replaced so we would not disturb the user with them?
If there are old binaries in "...\ProgramData\opencpn\plugins\" we may act in some way. But even if they are present and not enabled in config file they wouldn't be able to load and disturb OCPN. That's the main point as I see it?

Here are the log messages:
```
13:32:10.470 MESSAGE plugin_loader.cpp:258 Checking plugin candidate: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\statusbar_pi.dll
13:32:10.500 MESSAGE plugin_loader.cpp:1156 Plugin is compatible: false
13:32:10.500 MESSAGE plugin_loader.cpp:316 Incompatible plugin detected: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\statusbar_pi.dll
07:51:53.296 MESSAGE plugin_loader.cpp:316 Incompatible plugin detected: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\climatology_pi.dll
07:51:53.296 MESSAGE plugin_loader.cpp:258 Checking plugin candidate: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\gpsodometer_pi.dll
07:51:53.336 MESSAGE plugin_loader.cpp:1156 Plugin is compatible: false
07:51:53.336 MESSAGE plugin_loader.cpp:316 Incompatible plugin detected: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\gpsodometer_pi.dll
07:51:53.336 MESSAGE plugin_loader.cpp:258 Checking plugin candidate: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\nmeaconverter_pi.dll
07:51:53.366 MESSAGE plugin_loader.cpp:1156 Plugin is compatible: false
07:51:53.366 MESSAGE plugin_loader.cpp:316 Incompatible plugin detected: 
```

leamas, 2023-02-10:
> This code wouldn't have been tested enough before PR?

Actually tested it a lot. But there are corner cases when it comes to bad plugins....

leamas, 2023-02-10:
As for the other stuff: there are more fundamental changes unde way here, so it doesn't make much sense to look in to the details yet.

We are trying to address an issue when updating the catalog, which in some cases makes it impossible to  uninstall imported plugins. In other cases there is wrong metadata for an imported plugin. This is what make more drastic changes necessary. We need to sort this out first before getting into the details.

bdbcat, 2023-02-10:
Hakan...
I pushed a patch to disable the auto-timeout on messages of this type.  So at least now they can be read and thought about at user' s pleasure.
This may be a temporary workaround, until the changes stabilize.
Please try again, and report results.

Hakansv, 2023-02-10:
Now the pop-up messages are not timed-out. - OK
Still nothing happens when I click OK. OCPN is hanged badly.
What's with e.g. statusbar_pi.dll? It's not enabled.
[image]

bdbcat, 2023-02-14:
I have temporarily disabled the "incompatible plugin" message box.  There will be more discussion to come here.

leamas, 2023-02-14:
Yes. This is actually  long discussion which unfortunately has been in zulip private messages between me and Dave (just a coincidence, nothing private there). 

TL;DR This rabbit hole is deep.

I will try to somehow move the discussion from our  private space to a public zulip discussion, stay tuned.

leamas, 2023-02-14:
... done. Look into the  "Master 5.8.0 (was: comms)" zulip stream. There is new "Unloadable old plugins blues" topic. It's a lot of reading, not that well organized.

rgleason, 2023-02-14:
`In the case OCPN is disturbed by old plugin binaries not able to load we may inform the user and offer to delete the files. Since a OCPN plugin package installer or the Plugin Manager using Catalog/Import has once installed the file(s) the same OCPN program would also take responsibility to delete them.`

I like this idea, delete them after a positive response from a prompt.  Then the problem is taken care of.

Hakansv, 2023-02-14:
The code coursing the hang is now so far reverted so this issue is obsolete.

closed 2023-02-14 by Hakansv