OpenCPN/OpenCPN #3001: UI: Options-Charts-Vector

rgleason, 2023-02-11:
See the image

[image]

bc38, 2023-02-11:
scroll works for me
win11

rgleason, 2023-02-11:
Win 11  Scroll still does not work have tried many things.. opencpn_5.7.1+114.9715290_setup.exe

[opencpn.ini.txt](https://github.com/OpenCPN/OpenCPN/files/10714968/opencpn.ini.txt)
[opencpn.log.txt](https://github.com/OpenCPN/OpenCPN/files/10714969/opencpn.log.txt)
[image]

bdbcat, 2023-02-12:
Do you have your "display category" set to "User Standard" ?
Lower right hamburger...

rgleason, 2023-02-12:
I had tried setting the lower right hamburger "Display Catagory" setting  to various things, but perhaps did not set it at "User Standard."

Earlier I had tried to click on Options > Charts > Vector Charts  and "Reset to Standard" but that was greyed out and also did not work. 

Your suggestion worked. Thanks.  I had forgotten about this.

There is something about this arrangement and operation that is not terribly intuitive, because of the separation of the  "User Standard Objects"  from the lower right hamburger.   I think it might be worthwhile to consider moving this to have access from the lower right hamburger somehow  (later).

bdbcat, 2023-02-12:
We (and IMO)  consider "User Standard" to be an advanced feature, not generally required for normal navigation.  The reason for this is when watch changes on the bridge, incoming watch will see an easily recognized and familiar display setup. 
In fact, I think that IMO specs suggest that "User Standard" should not be automatically used (as specified in config file) from a normal startup, but must be directly selected by user on each program startup.   We don't go that far in OCPN. 
I've seen little post-it signs posted on commercial bridges: "DO NOT FIDDLE WITH THE ECDIS".

rgleason, 2023-02-12:
Ok, It is purposefully "Not Intuitive" for a good reason.  Perhaps a short note is needed in the manual. (to remind me).

closed 2023-02-12 by rgleason

rgleason, 2023-02-13:
Added a paragraph, but I need to do more work on Chars and Vector Display
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:charts:vector_charts#user_standard_objects