OpenCPN/OpenCPN #3003: UI: Size of Tide Icon with respect to Current Icon

rgleason, 2023-02-12:
I have tried adjusting the font from 9 up to 18 and it becomes apparent that there is too much difference in the respective sizes of tide icons (too small) and current icons (too big) IMHO which cannot be adjusted by changing the font size.
(I think P_dub is right.)

[image]

bdbcat, 2023-02-12:
Are you testing from current github master?

rgleason, 2023-02-12:
No, I am testing Beta 3

bdbcat, 2023-02-12:
OK, fine.  Just so we know.
Some user reports relate to current github master, and others to most recent Beta, and others something else entirely. It helps to know which is what, if you follow me.

rgleason, 2023-02-12:
Sorry, I've added the O version and Win 11 to all the issues that used that version.  My oversight.

Ptizef, 2023-02-12:
Dave, Rick
Window 11 tested with the current master (a few minutes ago just built)
I find the same things but with fewer differences. In fact, it has been improved by your previous changes
Now I can say that for a "reasonable"  chart object scale, and values font size for each, the tide icon is a bit too small and the current arrow a bit too large (or could be OK according to your taste)
In the shot attached, the chart object scale is zero and the font sizing is 10. My general display is scaled at 175 
JP
[image]

rgleason, 2023-02-13:
This appears to be much better.  I haven't tested myself, but I am closing.

closed 2023-02-13 by rgleason