OpenCPN/OpenCPN #3004: Sector light rings flash on/off in course-up lookahead mode

transmitterdan, 2023-02-12:
**Describe the bug**
Light rings flash on/off

**To Reproduce**
Steps to reproduce the behavior:
1. View ENC chart with sector lights
2. Set course-up mode
3. Set look-ahead mode
4. Yellow rings flash as chart automatically pans/rotates but in other modes they do not

**Expected behavior**
Rings should not flash

**Screenshots**
https://user-images.githubusercontent.com/10525699/218314091-716c2d39-6c79-4e7c-bd26-a82e4cc0222b.mp4

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Version 11

transmitterdan, 2023-02-12:
I guess this might be related.  On some platforms (Arm-Pi Jammy) the busy spinnger is constantly flashing on/off too.

Hakansv, 2023-02-13:
The busy spinner is there on my RPis as well using head up mode. And CPU load about two times more.
(Present master and o-charts.)

bdbcat, 2023-02-18:
Cannot reproduce.
Anyone else?

transmitterdan, 2023-02-19:
I am running a slow Linux machine running x86_64 Jammy. It's slow because it's a 4 thread WSL setup.
```
transmitterdan@DESKTOP-2B2E08B:~$ cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.1 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
transmitterdan@DESKTOP-2B2E08B:~$
```
I think this is a busy spinner call during chart rotation/translation that is invisible on fast machines but not on slow machines.

bdbcat, 2023-02-19:
Thread drift.
Anyone else able to reproduce the flashing sector lights?

rgleason, 2023-02-19:
In Windows 11  Opencpn from master recent build. In course up, with lookahead mode, CM93, runnning VDR with Hartmut-AN-Markermeer.txt  and I see light sectors, that are jiggling a bit as the chart turns back and forth, these sectors are being redraw each time the chart turns.  On a slow machine this would probably appear to be flashing?

rgleason, 2023-02-20:
Also, after awhile even the jiggle disappears, probably because it is cached and reused for each redraw?

transmitterdan, 2023-02-22:
I think this is related to wx versions.  The current CI build using the canned wx libs does not appear to do it.  Closing. I'll work it against recent wx master and see what I can find.

closed 2023-02-22 by transmitterdan