OpenCPN/OpenCPN #3006: UI: Options > Plugins  - New Message about Incompatible Plugins

rgleason, 2023-02-13:
This new message is a pain. What good is it to a user?  
I get 15 messages stacked up and have to wait a long time before I can click 15 times to get rid of the message.

What is it's purpose?  I am not going to do anything about it. The plugins are imcompatible and will not show. It is a complete bother and waste of time.

I would suggest that you collect all the plgins that are not compatible and write the list to the log file once, as a comprehensive list.

If that is no suitable, present the notice with a single popup at the end of loading with a complete list of incompatible plugins, and a brief note suggetsting what the USER should do about it.
[image]

bdbcat, 2023-02-13:
Rick...
This dialog, and consequences, are a work -in-process.
Rest assured that this will be fixed before next Beta.

Hakansv, 2023-02-13:
Rick..
If you don't mind there's already an [issue](https://github.com/OpenCPN/OpenCPN/issues/2997) about this.
Not that my are more important but we may try to not flood the issue system?

rgleason, 2023-02-13:
Did not know that. Thanks. closing

closed 2023-02-13 by rgleason