OpenCPN/OpenCPN #3007: GRIB Plugin - Found CPN crash with wind particles enabled

bc38, 2023-02-13:
Found CPN crash with wind particles enabled and GRIB data on screen. There is also a twitching of the bar at the top of the screen.
Recently, some improvements have been made to the Grib plugin, but the plugin version does not change

rgleason, 2023-02-14:
win 11  recent build of master  appears to be working ok.
[image]

bc38, 2023-02-14:
rgleason, 
i checked on ENC S-57 cards
checked on the CM93 map, particle map "5" when working with zoom +/- no crash! changed particle value to max 10 and got crash.
I will record a video later

bc38, 2023-02-14:
try switching zoom +/-

rgleason, 2023-02-14:
While running OpenCPN from Visual Studio, with the grib plugin and the NZ cyclone +/- do not seem to crash it either.   Win 11 with very recent Opencpn master.

bc38, 2023-02-14:
https://www.dropbox.com/s/indscmohvjtrss5/GRIB.mp4?dl=0
Video! GeForce GTX 1070 Ti

Hakansv, 2023-02-14:
bc38.
Dave has recently made some updates regarding NVIDIA and GeForce. Until you have a possibility to test if the new code may help, you may try without OpenGL?

bc38, 2023-02-14:
> bc38. Dave has recently made some updates regarding NVIDIA and GeForce. Until you have a possibility to test if the new code will help, you may try without OpenGL?

without OGL all is well there is no crash
BUT without OGL, the particles and the weather map in color are simply not displayed!
GRIB plugin needs graphics optimization

bc38, 2023-02-25:
Another manifestation of a bug on one of the scales, mainly on ENC S57 charts
[image]

rgleason, 2023-02-25:
Well, one good thing is that this is an important plugin, which can be fixed more readily later. ...an important plugin though.

bc38, 2023-03-03:
I noticed that the problem with particles only appears on GRIB ICON files where the wind resolution is higher than that of GFS.
Attached is the Grib file (.zip del). Check if the problems described above are reproduced?

[ICON20230303024000277.grb.zip](https://github.com/OpenCPN/OpenCPN/files/10877748/ICON20230303024000277.grb.zip)

rgleason, 2023-03-03:
What a beautiful[ lake ozero baikal](https://duckduckgo.com/?q=ozero+baikal&t=newext&atb=v362-1&iax=images&ia=images&iai=http%3A%2F%2Fwww.fotosselect.ru%2Fwp-content%2Fuploads%2F2015%2F05%2Fozero-baikal-samoe-glubokoe-v-mire.jpg)

Bc38 I downloaded and tested it with Particles checked.  It used a relatively large amount of memory, but did not crash Opencpn. 
Could it be that it is also related the to charts being used, or perhaps the computer?  It is puzzling.  Sorry.
[image]

[image]

rgleason, 2023-03-03:
Is this a fix by Ptzef?  https://github.com/OpenCPN/OpenCPN/commit/23868eed2e71e37aca23f995cac259e0a758bcad

bc38, 2023-03-04:
I tested on a desktop PC with an AMD Ryzen processor. checked on a laptop where there is also an AMD Ryzen processor and a Radeon video
crash when turning on particles, reproduced on a laptop.
tested on an old mini pc with Intel Celeron and Intel HD video
and I didn't find any problems with crashing on particles, even with the maximum value of particles.
What is the problem in the AMD processor?

rgleason, 2023-03-04:
Lets wait and see if ptzef's work fixes it when it gets merged?
https://github.com/OpenCPN/OpenCPN/pull/3051/commits/c5acad52c1e02f99e803fd9c84d7eede0834d858

leamas, 2023-03-04:
@bc38: Any chance you could provide a stack trace?

bc38, 2023-03-04:
> @bc38: Есть ли шанс, что вы могли бы предоставить трассировку стека?

If you tell me how to do it, I will try

leamas, 2023-03-04:
You are on windows, right? I don't know that, others do,  we need to document it for sure.

@rasbats^ ideas?

bc38, 2023-03-04:
yes windows 11 and have 10

Rasbats, 2023-03-04:
Intel processor. Unable to reproduce the crash.

leamas, 2023-03-04:
@Rasbats yes but have you ideas about windows stacktraces?

rgleason, 2023-03-04:
Alec I have not had any crashes, so no stacktraces are available.  Is there something else that would help?

rgleason, 2023-03-04:
bc38 can you download [glview](http://www.realtech-vr.com/home/glview) and run some tests for us with the Radeon?  ..and post the results here?

bc38, 2023-03-05:
[Desktop.zip](https://github.com/OpenCPN/OpenCPN/files/10890721/Desktop.zip)
[Laptop.zip](https://github.com/OpenCPN/OpenCPN/files/10890726/Laptop.zip)

Rasbats, 2023-03-05:
@bc38 : A crash report should have been generated in _ProgramData/opencpn/CrashReports_. Could you post the latest of these and I can take a look with _WinDbg_?

bc38, 2023-03-05:
[9a40cb39-a8ab-48cd-8aba-3700fb9064f8.zip](https://github.com/OpenCPN/OpenCPN/files/10891542/9a40cb39-a8ab-48cd-8aba-3700fb9064f8.zip)

AMD Ryzen 5
GeForce 1070 ti
Win11

Rasbats, 2023-03-05:
Thanks. WinDbg shows this trace:

```
STACK_COMMAND:  ~0s; .ecxr ; kb

SYMBOL_NAME:  opengl32!glDrawArrays+25

MODULE_NAME: opengl32

IMAGE_NAME:  opengl32.dll

FAILURE_BUCKET_ID:  INVALID_POINTER_READ_c0000005_opengl32.dll!glDrawArrays

OSPLATFORM_TYPE:  x86

OSNAME:  Windows 10

IMAGE_VERSION:  6.2.22621.608

FAILURE_ID_HASH:  {09c109b2-2af4-1d30-9c41-02e56cb3641b}

Followup:     MachineOwner
---------

```

rgleason, 2023-03-05:
bc38 what is the result of the opengl testing?  Compatible up to what version?  Any failures?

rgleason, 2023-03-05:
@Rasbats Did you install the Microsoft store version?  any particular selections needed that you can remember?

bc38, 2023-03-05:
> bc38 what is the result of the opengl testing? Compatible up to what version? Any failures?

all tests from version 3 to 4.5 passed without problems

bdbcat, 2023-03-05:
FWIW, I am currently working on GRIB_pi, to update rendering methods. following OCPN core OpenGL/GLSL architecture.
It will take a couple of days....

bdbcat, 2023-03-06:
Hello all...
I have adapted GRIB_PI to use more modern OpenGL methods.
Please test this:
[https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.7.1+257.fdcb3ff_setup.exe](url)
Good Luck

Hakansv, 2023-03-06:
Win10.
Works fine. More clear icons and colors.

rgleason, 2023-03-06:
Pulled current master and built. It seems to be working well on Win 11

bdbcat, 2023-03-07:
But is it working for bc38?

bc38, 2023-03-07:
> But is it working for bc38?

I checked on two devices where the crash was reproduced. Miracle! There is no collapse even at the maximum particle value.
Thank you all for your response to the problem!
I think we can close!

Off topic
- On a Geforce 1070ti desktop PC, the top of the screen twitches when displaying particles. See video
https://user-images.githubusercontent.com/7856184/223437192-2039030f-9503-4fce-81bf-fd42063ed029.mp4
- Icons are small in scaled screen mode more than 100%
- It's time to change the version of the Grib plugin

closed 2023-03-08 by bc38