OpenCPN/OpenCPN #3008: Very recent master build Exceptiobn

rgleason, 2023-02-14:
Win11 current master, vstudio build, exceptions
Just updated Visual Studio also. Got these exceptions, won't build Opencpn yet. 
Sorry this is not more organized. Will try more later.

[exception.txt](https://github.com/OpenCPN/OpenCPN/files/10735910/exception.txt)
[image]

[image]
[image]
[image]
[image]

transmitterdan, 2023-02-15:
I have latest VS2022 and don't see this. I suggest you do a clean and rebuild all the sources. This is definitely not a bug in OpenCPN. Also, be sure you are using CMake correctly to create the project. Look at the CI build scripts for hints how to run CMake at the command line for VS2022.

rgleason, 2023-02-15:
That seems to have fixed building Opencpn. Thanks.

closed 2023-02-15 by rgleason