OpenCPN/OpenCPN #3011: Unloadable plugin library files wreaks havoc

leamas, 2023-02-15:
**Describe the bug**
In certain situations, notably after a ABI breaking catalog update, existing plugin files cannot be loaded.

**To Reproduce**
As described in #2997

**Expected behavior**
User dialog which makes it possible to remove plugin containing the library or just the library

**Desktop (please complete the following information if applicable):**
This is a problem on all platforms

**Additional context**

This has been discussed in length at [zulip](https://leamas.github.io/zulip-archive/stream/332168-Master---5.2E8.2E0--.28was.3A-comms.29/topic/Unloadable.20old.20plugins.20blues.html)

This bug represents the handling of existing, unloadable plugins discussed here. Another bug will be filed for sanitizing the problems associated with ABI breaking updates as of the zulip discussion

rgleason, 2023-02-15:
I read much if the Zulip dialog,
1. Since I tend to make new version numbers some of these problems are alleviated.
2. One wrinkle about this is we are assuming users don't use two different versions of OpenCPN at an ABI break, and we are solving the problem in the simplest way, assuming the User intends to move forward with the new version.  I think this is reasonable, if they decide to go back to the earlier version they can simply reload the plugins.

rgleason, 2023-02-15:
This suggestion by Alec seems reasonable to me: It does a lot for the user.

> The other case is about making sure that ABI breaking updates of opencpn works in a sane way Could be:
> 
>     On startup, opencpn checks for installed plugins with incompatible ABI:s (in parallel thread, not delaying start up This test should be cached to not delay normal start up).
>     If such are found and the catalog is too old to have new, wx32 entries user get a chance to update the catalog (modal dialog).
>     All incompatible plugins are uninstalled.
>     For plugins uninstalled, we look for compatible entries with same name in the catalog. If existing, this is installed. If there are multiple matches the one with highest version number is used.
>     Possibly, a dialog informs user about update actions ("The following plugins have been updated")
>

leamas, 2023-02-15:
@rgleason  : none of these remarks are related to this issue. As said in the top comment there will be another bug filed later on this

leamas, 2023-03-01:
Closed by #3024

closed 2023-03-01 by leamas