OpenCPN/OpenCPN #3014: OCPN not functioning when launched from Terminal on MacOS

antipole2, 2023-02-18:
**Describe the bug**

When OCPN v5.7 is launched from Terminal on MacOS (Venturer), it does not completely launch.  The toolbar does not appear.

Using Beta #3

**To Reproduce**

1. Open Terminal
2. $ /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
3. Then click on canvas to give it focus

No toolbar!

**This means it is not possible to debug OCPN**

**Desktop (please complete the following information if applicable):**
 - OS: MacOS 13.1 Ventura

**Additional context**

This issue occurs on both Intel and M2 Macs

The same procedure on OCPN v5.6.2 works fine.

nohal, 2023-02-18:
This is exactly how I start OpenCPN many times a day, without experiencing this issue. The only difference is my 2015 Mac can't be updated to Ventura. We will probably need some more info to do something about this.

antipole2, 2023-02-18:
I have just fired up an old MacBook running Catalina 10.5.7 and used terminal to launch the same OCPN.
It works OK.
So it is somehow linked to post-Catalina changes.

nohal, 2023-02-18:
It works for me also on latest Moterey, which is as far as I'm able to update this machine, so we still probably need you to isolate what it actually is that is not working. To start with, try to rename your configuration file before starting OpenCPN.

antipole2, 2023-02-18:
Already been there.  Renamed OpenCPN folder in Preferences and checked no OCPN folder in Applications Support.
Have just done it again to double check.
No luck.  So problem occurs with completely clean start.

bcn2, 2023-02-18:
Can confirm the bug on our macMini M1/macOS 13.1:

Mac-mini:~ renate$ /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
 SENC: nCPU: 8    m_max_jobs :7
Generating RSA key...
Generating x509 certificate...
Writing key and certificate to disk...
Success!
Local IPv6 address: fd00::858:72a0:ac85:536f
Local IPv4 address: 192.168._xxxxx_
Opened 1 sockets for mDNS service
Service mDNS: opencpn-object-control-service.:8000
Hostname: Mac-mini
Sending announce
2042d07a 3 mongoose.c:3461:mg_listen    1 0x13 https://0.0.0.0:8443
Mac-mini:~ renate$

nohal, 2023-02-27:
Anyone able to provide some (Like stack trace in the "hanging" state)info that we could use to track this down?

antipole2, 2023-02-28:
Issue still exists with OCPN5.7.1 beta4

bdbcat, 2023-02-28:
Anything in the logfile?

bdbcat, 2023-02-28:
I start O from CLI like this:

`$ open /Applications/OpenCPN.app`

Any difference if you try that?

bcn2, 2023-02-28:
Either with the debugger:
 lldb /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
 r
or directly
 /Applications/OpenCPN.app/Contents/MacOS/OpenCPN 

however - using your command open...  all is fine.

antipole2, 2023-03-06:
I believe this issue is a wx3.2 byproduct.
See my experiments detailed  in #3044.
When the focus moves away from OCPN the toolbar is not displayed.  Bringing the focus back to OCPN displays the toolbar again.  So if you launch from Terminal in such a way that the focus is not on OCPN, the toolbar will not be displayed.
Now if a modal window has been displayed as part of OCPN starting up, it is not possible to bring OCPN into focus until the modal window has been dismissed.
In multi-screen or multi workspace situations, the modal window may not be on the same screen as OCPN.  It may not be obvious. You have to hunt for fit.
Further, I have seen the toolbar appear in a different space to OCPN.  Have to hunt for it.  Having visited the space where the toolbar is lurking, return to OCPN and the toolbar appears with OCPN.

nohal, 2023-04-30:
Moving to 5.10. I can't reproduce this on the usual single display setup I have available and we will certainly not be able to address this for the 5.8.x maintenance releases.

nohal, 2024-02-13:
I am not able to reproduce any evidently broken behavior even with two monitors connected. Please create a new issue with exact steps to reproduce if there is still something broken with current CI builds. Emphasis on the CI builds as not even wx 3.2.4 as released works correctly on Sonoma and we are applying a few patches over it.

closed 2024-02-13 by nohal