OpenCPN/OpenCPN #3019: macOS: missing labels on sliders of Options

bcn2, 2023-02-19:
**Describe the bug**
At macOS the numerical labels of the sliders in the Option menues are missing. Nothing new from the beta. Has been there on 5.6.2 as well

**Screenshots**
[image]
[image]

**Desktop (please complete the following information if applicable):**
 - OS: macOS

bcn2, 2023-02-26:
After the commit some advance:
ok, when there is NO lateral slider of the Window for scrolling.
In the moment one touches the slider to scroll, the UI sliders get compressed
[image]
[image]
[image]
[image]

bdbcat, 2023-02-26:
That is a wxWidgets bug.  Very strange, only seen on MacOS.
Might be fixed in wx3.2.2.1, not tested yet.
Struggling to find a workaround.
As far as I can tell, this bug existed in O562 also.

nohal, 2023-02-26:
This wx bug exists for many years, that's why we disabled the slider labels on macOS long time ago.

bdbcat, 2023-02-26:
My fiddly plan for workaround is to process scroll events. and resize the sliders (back to their default size) dynamically.   Sigh....

leamas, 2023-02-27:
Is there an upstream bug?

bdbcat, 2023-02-27:
Probably, but not deeply investigated.
We hoped it would have been magically fixed by wx32.  Nope.
Finding and fixing the upstream is the "right" way to fix this....

bdbcat, 2023-02-28:
Pavel....
I pushed a partial correction to github master.
Please validate, and tweak if inspired.  Not one of my proudest efforts...

nohal, 2023-02-28:
TBH, I would probably not do it, the labels are IMO not worth it. It kind of works for on the UI tab, but the shape changing is pretty annoying... Will try to sleep on it and see if it can be forced to behave at least somewhat better.

bdbcat, 2023-02-28:
It is strange behavior.  I would bet that there is some scrambling of object style flags, somewhere in wx...

nohal, 2023-02-28:
Probably better than trying to work it around in our code...

nohal, 2023-02-28:
The upstream issue is https://github.com/wxWidgets/wxWidgets/issues/16302

leamas, 2023-02-28:
If we do it, we should IMHO apply a patch to the original sources in the build process rather than just patching them directly. This makes it much easier to maintain when wxwidgets makes new releases, etc.

bcn2, 2023-02-28:
From Pavel's link to the original bug, that's from 2014. 
And yes, repair upstream. If we can't show values for the sliders, that's not a drama. Just ugly.

bdbcat, 2023-02-28:
"apply a patch to the original sources "
Actually, we do not build wx from sources as part of the CI now.  Perhaps we should, but that is a different discussion.
So, while I agree in principal, this might not be the idea time to do that.  It could be a nice exercise for "between-release" time.

bdbcat, 2023-03-01:
Updated MacOS build with locally patched wx322 available here:
[https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.7.1.pkg](url)

leamas, 2023-03-01:
For the record: could you please attach the patch(es) applied here?

bcn2, 2023-03-01:
Patch working ok here.

bdbcat, 2023-03-02:
[0001-Correct-slider-distortion-when-used-in-sizer-MacOS.patch.txt](https://github.com/OpenCPN/OpenCPN/files/10866969/0001-Correct-slider-distortion-when-used-in-sizer-MacOS.patch.txt)

bcn2, 2023-03-05:
Shall we close the issue?

bdbcat, 2023-03-05:
Good for me.

closed 2023-03-05 by bcn2