OpenCPN/OpenCPN #3021: PI: Import Plugin  is broken -- Every plugin is incompatible.

rgleason, 2023-02-19:
**Describe the bug**
Import Plugin is totally broken.

[image]
[image]

bdbcat, 2023-02-20:
Going to need some more specific information.

Please start with one plugin, so we can reproduce your experience.
What plugin?  Where is the URL for the plugin.

bdbcat, 2023-02-20:
Ping?

rgleason, 2023-02-20:
Sorry, meeting with Contractor soon. More this afternoon.

rgleason, 2023-02-20:
It has happened on Polar_pi and Autopilot_Raymarine_pi  which I tried earlier.
Now it has happened on weather_routing_pi  using the local build 
[weather_routing_pi-1.13.49.0-msvc-x86_64-10.0.22000.tar.gz](https://github.com/OpenCPN/OpenCPN/files/10785880/weather_routing_pi-1.13.49.0-msvc-x86_64-10.0.22000.tar.gz)

https://github.com/rgleason/weather_routing_pi

bdbcat, 2023-02-20:
https://github.com/OpenCPN/OpenCPN/files/10785880/weather_routing_pi-1.13.49.0-msvc-x86_64-10.0.22000.tar.gz

This tarball/plugin is built for wx3.1.
So of course it is incompatible with O571, which uses wx32
Expected result.

rgleason, 2023-02-20:
Dave I just rebuilt it, locally, using my bld.bat file:
```
cd build
cmake -T v143 -A Win32 ..
cmake --build . --target package --config relwithdebinfo >output.txt
bash ./cloudsmith-upload.sh
```
which in addition to the tar.gz above, creates an xml file which I include below.

[weather_routing_pi-1.13.51.0-msvc-x86_64-10.0.22621.xml.txt](https://github.com/OpenCPN/OpenCPN/files/10788399/weather_routing_pi-1.13.51.0-msvc-x86_64-10.0.22621.xml.txt)

This file has 
```
<target>msvc-wx32</target>
<build-target>msvc</build-target>
<build-gtk></build-gtk>
<target-version>10.0.22621</target-version>
<target-arch>x86_64</target-arch>
```

bdbcat, 2023-02-21:
Rick...
The metadata file embedded in the tarball is the controlling factor.  It looks like this:
<target>msvc</target>
<build-target>msvc</build-target>
<build-gtk></build-gtk>
<target-version>10.0.22000</target-version>
<target-arch>x86_64</target-arch>

rgleason, 2023-02-21:
I thought the <target> was controlling, and that is msvc-wx32 !!!

`<target>msvc-wx32</target>`

What tried next was just renaming the fill to  weather_routing_pi-1.13.51.0-msvc-wx32-x86_64-10.0.22621.tar.gz
and tried importing and it accepted the file and imported!

PS: You looked in the tar.gz and the metadata file does not have <target>msvc-wx32 ???
Because the metadata.xml in the build directory does have that.

rgleason, 2023-02-21:
I think the problem I am having is just that the Testplugin, cmake setup to build locally, does not name the files properly, because the File manager seems to be looking at the file name itself, rather than the metadata.xml file contained within the  tar.gz file.

I think this is actually an error,  because Alec has said in the past that it does not matter what the name of the file is.

bdbcat, 2023-02-21:
Rick...
In the file URL 
https://github.com/OpenCPN/OpenCPN/files/10785880/weather_routing_pi-1.13.49.0-msvc-x86_64-10.0.22000.tar.gz
The embedded (i.e. controlling) XML does indeed say:
target>msvc

This one:
https://github.com/OpenCPN/OpenCPN/files/10788625/weather_routing_pi-1.13.51.0-msvc-wx32-x86_64-10.0.22621.tar.gz
has msvc-wx32, seems correct.  Probably will import successfully.

I cannot account for your build scripts, sorry.

It still does not matter what the import file name is, nor what the XML file name is.  Only matters is the content.
During import O will check the content of the embedded XML.

rgleason, 2023-02-21:
Yes, when I rename this  file to  weather_routing_pi-1.13.51.0-msvc-x86_64-10.0.22621.tar.gz
and try import, it does work.

Ok, so I am going to have to watch what is in the metadata.xml file.
Sorry about that.  It looks like PM is working.

closed 2023-02-21 by rgleason