OpenCPN/OpenCPN #3023: RFE: Grib_pi  - Add ECMWF to grib plugin email requests

rgleason, 2023-02-21:
See Cruisers Forum  LeaseOnLife post https://www.cruisersforum.com/forums/f134/opencpn-version-5-7-1-beta-test-3-a-272640.html#post3745635

some ECMWF data are available since March 2022 via  [query@saildocs.com](mailto:query@saildocs.com):
WIND, MSLP, HGT500, TEMP (2m air temp) and WAVES

Having another global model besides GFS to compare forecasts / trends, beyond a couple days, is valuable.

From saildocs::

NOTICE: From 8-Mar 2022, ECMWF "Essential" is replaced with ECMWF "Open Data",  now available without charge. Parameters are limited compared to full  ECMWF access and the data is delayed about one hour.   Global forecast  data is available from 0-144 hours at 3-hour increments, and 150-240  hours at 6-hour increments. The data grid is 0.4-deg, and available  parameters at this time are WIND,  MSLP, HGT500, TEMP (2m air temp) and WAVES (sig wave height). Data is  updated every 12 hours, and available by 8 hours after the synoptic time  e.g. 08:00 utc for the 00z model-run. The request code remains "ECMWF".    Usage is subject to the ECMWF terms and conditions, refer to "Gribmodels" (https://saildocs.com/gribmodels).

Some ECMWF data are available since March 2022 via [query@saildocs.com](mailto:query@saildocs.com):
[WIND](https://www.cruisersforum.com/forums/tags/wind.html), MSLP, HGT500, TEMP (2m air temp) and WAVES

Having another global model besides GFS to compare forecasts / trends, beyond a couple days, is valuable.

Usage is subject to the ECMWF terms and conditions, refer to "Gribmodels" (https://saildocs.com/gribmodels).

bc38, 2023-03-08:
implemented! can be closed

rgleason, 2023-03-08:
Yes, ECMWF is working. Requested, recd and tried.  Thanks.

closed 2023-03-08 by rgleason