OpenCPN/OpenCPN #3025: cairo.dll missing from OpenCPN 5.7.1 installation - results in SVG problems for Windows

Rasbats, 2023-02-25:
**Describe the bug**
Plugin svg icons using alphanumeric text e.g. {JS} do not show in the toolbar or list of plugins.

**To Reproduce**
Steps to reproduce the behavior:
Install JavaScript_pi on Windows (available for wx32 in the alpha Cloudsmith repo). Instead of icons a black square is shown. These icons are made with text characters.

**Expected behavior**
Icons should appear as other plugin icons.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
Windows 10

**Additional context**
Workaround is not to use alphanumeric text but this adds complexity when building icons.
This refers:
[https://forums.wxwidgets.org/viewtopic.php?t=44708](https://forums.wxwidgets.org/viewtopic.php?t=44708)

nohal, 2023-02-25:
We are not using Cairo for SVG rendering anymore since we switched to `wxBitmapBundle` available in wx3.2. If the SVGs used by this plugin are problematic, they need to be fixed, there is no way back to switching the rendering backend, it caused too many problems and was very hard to maintain.

Rasbats, 2023-02-25:
Thanks for the quick reply. I will advise Tony Voss.

closed 2023-02-25 by Rasbats