OpenCPN/OpenCPN #3028: Reduced details at small scale.

RooieDirk, 2023-02-27:
With the 'reduce additional at small scale' checked the behavour at zooming in was that you would see appear buoys first at a smaller size and with more zooming in they 'grew' to normal size.
No matter what I try, it doesn't work anymore. OpenGL on/off additional reduce on/off charttext on/of, no difference. The bupys keep popping up at full size including the text belonging to the buoys.
My system
Linux (22-4), GTK3 and wxWidgets 3.3

RooieDirk, 2023-02-27:
Yes I admit I do have a custom build wxWidgets, the rest is standard Kubuntu22-4

leamas, 2023-02-27:
Not much standard here. There is no official support for neither Kubuntu nor custom wxWidgets builds. 

It might be useful if you could reproduce this on a supported platform.

RooieDirk, 2023-02-27:
I might have to change this form a bug int a future request. What I mend is still working but only if the detail slider is set to a positive value . I realy like the effect so it would be nice to have it always on.
See screenshot movie. Detail slider to 0 -> the buoys will popup from nothing to all. Slider to -1 and the buoys will come up small and growing while zooming more.

https://user-images.githubusercontent.com/2668258/221630302-99f80ebf-b234-4dc2-9380-345ee7e02eff.mp4

bdbcat, 2023-02-28:
I'd call this a new Feature Request.  It is a bit late in the O58 test cycle, but might be interesting in the future.
Thanks
Dave

erik-meuwese-topicus, 2023-07-23:
Is this not a chart issue? I get the same behaviour in OpenCPN 5.8.4 on Windows 11 with `1R7WZ003` in the set `Nederland (excl. Zeeland) 2023-01-18` from Rijkswaterstaat. I don't get this behaviour with `1R7WAD08` in the set `Waddenzee met Diepte 2023 - week 29` from Rijkswaterstaat

`1R7WAD08` in the set `Waddenzee met Diepte 2023 - week 29` :
https://github.com/OpenCPN/OpenCPN/assets/79573173/3b5f5f8e-2e9e-4a3a-8211-10488d2b7853

`1R7WZ003` in the set `Nederland (excl. Zeeland) 2023-01-18` :
https://github.com/OpenCPN/OpenCPN/assets/79573173/3a8b0dc6-ddb1-4d52-a209-bd9f33258303

bdbcat, 2025-03-14:
The function desired is present, but only with zoom scale > 1, as noted.
Does not at all depend on "Additional detail reduction at small scale" setting, and never has.
Converting to "Enhancement", for further discussion if desired.