OpenCPN/OpenCPN #3034: Mac small issue: Inconsistency between Terminal and System Settings Output - MacOS issue?

rgleason, 2023-02-28:
Note: This appears to be a MacOS issue not related to OpenCPN.  If so, please advise and I will close this.

**Describe the bug**
Mathais W. reports:

> A friend helped and pushed me to get OpenCPN connected to my Mac at last - he had been pushing for a year ;). I was following this nice guidance for using the YDNU02 adapter, which looks straightforward enough...
> 
> https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:supplementary_hardware:seatalk_nmea
> 
> Unfortunately, it did not work from the start. I then started digging a bit and alas, eventually I found out that the information presented in the “easy approach” via System Settings” on a Mac is not the same as when using the Terminal application. See the screenshot attached. The System Settings is somehow cutting off the last digit, it seems.
> 
> So, not sure whether this is something special for my case, but perhaps worth mentioning on the OpenCPN web page to also look at the Terminal application using the ls command as seen to double check, if it does not work out of the box.
> 
Inconsistency between Terminal and System Settings Output.png
[image]

leamas, 2023-02-28:
This is no inconsistency. The System Settings reflects the USB device properties, as can be seen  on linux using lsusb. MacOS seems to have some more complicated CLI commands displaying the same info.

OTOH, the files in /dev are, well, files.  There is a relation between the two, the filename is sort of a handle to the USB device. But as I see it, the possible problem here might be about documentation.

rgleason, 2023-02-28:
Thanks. Added to Sup Hardware here https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:supplementary_hardware:seatalk_nmea#macos_notes

closed 2023-02-28 by rgleason