OpenCPN/OpenCPN #3038: Beta #3 booms out when click on Options > Plugin

rgleason, 2023-03-01:
**Describe the bug**
1. Beta #3  O crash when click on Options > Plugin  see attached log
2. Very recent master, built and VS debug, results in immediate break. https://github.com/jongough/ocpn_draw_pi/issues/517#issuecomment-1449052306

[opencpn.log.txt](https://github.com/OpenCPN/OpenCPN/files/10856023/opencpn.log.txt)

rgleason, 2023-03-01:
Stops when I delete  %localappdata%\opencpn\plugins\ocpn_draw.dll

Then reinstalled ocpn_draw.dll and it started crashing again.

Then deleted   %localappdata%\opencpn\plugins\ocpn_draw.dll

Remove the ".zip" please
[ocpn_draw_pi.dll.zip](https://github.com/OpenCPN/OpenCPN/files/10856086/ocpn_draw_pi.dll.zip)

rgleason, 2023-03-01:
Have reloaded ocpn_draw 1.8. 21  and it seems to be working ok now.
I think there is possibly some glitch in the PM that caused this, but perhaps the sreenshots of the break on jongough, linked to above will help identify.

bdbcat, 2023-03-01:
Please clarify.
1:    1.8.20 crashes.
...fix some bug...
2:   1.8.21 works ok.

Seems normal progression to me. So what is there to analyze?

Hakansv, 2023-03-01:
I can verify the initial Plugin Manager, PM, failure. Despite in safe mode O crash on a bad plugin.
Method:

1. Open PM and switch to and update catalog Beta. 
2. ocpn_draw, OD, was ready to update and I did so.
3. OCPN crash. Now started O from VS. Answered Yes to use safe mode wo plugins.
4. When O is now running wo plugins I opened PM in order to take out the offending OD. 
5. OCPN crash. VS debug brakes on loading a NULL bitmap from OD! PM row: 366 
Now I haven't build OD so no symbols to get further details.
6. Tried a new run to confirm I really used safe mode. Answered again Yes to not load plugins. Still crash in PM row 366. NULL bitmap. Call stack says only ocpn_draw ... no symbols.
7. Since I used safe mode it was not expected PM should crash on a plugin having some bitmap failure.
Now I had to delete the OD dll from \plugins to be able to overcome the crash and enable other plugins.

leamas, 2023-03-01:
@Hakansv : Could you please attach a --loglevel debug logfile from a crashing session?

rgleason, 2023-03-01:
Please  note that the crash first occurred using VS with very recently pulled master, see the screenshots above which occurred as soon as I hit Options > Plugins.

Unfortunately I then downloaded Beta #3 instead of #4, so those results are not germaine.

leamas, 2023-03-01:
I can see the problem in plugin_loader, and it's an old one. We basically try to load a  plugin before the test for safe_mode, so the only strange thing is that this code has been with us for a while.

However, late changes might have made things worse since we now load also disabled plugins to discover if they are faulty.

I'm on it, but would appreciate a logfile.

leamas, 2023-03-01:
That i crashes in normal mode is unavaidable, but we should fix safe mode.

Hakansv, 2023-03-01:
Yes, agree.

Hakansv, 2023-03-01:
> and it's an old one. We basically try to load a plugin before the test for safe_mode,

If you're able to fix this I've a user crash on safe mode not understood so far. It may then be a bad plugin although in safe mode. I really look forward to retest that one.
Thanks.

leamas, 2023-03-01:
This is a can of worms. We start loading plugins before we process the `-safe-mode` command line parameter with all sorts of interesting results. Need lunch

leamas, 2023-03-01:
@Hakansv : Can you try this patch?
```
diff --git a/src/ocpn_app.cpp b/src/ocpn_app.cpp
index 7aa9bfbb7..5d6709dc6 100644
--- a/src/ocpn_app.cpp
+++ b/src/ocpn_app.cpp
@@ -1063,8 +1063,6 @@ bool MyApp::OnInit() {
 #endif
 
   GpxDocument::SeedRandom();
-  safe_mode::set_mode(false);
-
   last_own_ship_sog_cog_calc_ts = wxInvalidDateTime;
 
 #if defined(__WXGTK__) && defined(ocpnUSE_GLES) && defined(__ARM_ARCH)
diff --git a/src/plugin_loader.cpp b/src/plugin_loader.cpp
index c1fa24a8a..6871586e9 100644
--- a/src/plugin_loader.cpp
+++ b/src/plugin_loader.cpp
@@ -98,6 +98,18 @@ PlugInContainer::PlugInContainer() {
   m_api_version = 0;
 }
 
+/** Return true if path "seems" to contain a system plugin */
+static bool IsSystemPlugin(const std::string& path) {
+  static const std::vector<std::string> SysPlugins = {
+    "chartdldr_pi", "wmm_pi", "dashboard_pi", "grib_pi" };
+
+  const std::string lc_path = ocpn::tolower(path);
+  for (const auto& p : SysPlugins) {
+    if (lc_path.find(p) != std::string::npos) return true;
+  }
+  return false;
+}
+
 SemanticVersion PlugInContainer::GetVersion() {
   if (m_ManagedMetadata.version.size()) {
     return SemanticVersion::parse(m_ManagedMetadata.version);
@@ -293,6 +305,10 @@ bool PluginLoader::LoadPluginCandidate(wxString file_name, bool load_enabled) {
   }
 
   if (loaded) return true;
+  if (!IsSystemPlugin(file_name.ToStdString()) && safe_mode::get_mode()) {
+    DEBUG_LOG << "Skipping plugin " <<  file_name << " in safe mode";
+    return false;
+  }
 
   bool b_compat = CheckPluginCompatibility(file_name);
 

```

Hakansv, 2023-03-01:
@leamas 
Yes, half way.. Good!
No I can in safe mode enter PM without crash.
But the offending plugin ocpn_draw is not listed in PM, see pictures below, so not possible to uninstall. 
So when running in normal mode next time O will crash again since the ocpn_draw_pi.dll is still present.
If the user in safe would detects ocpn_draw is missing in the PM's list he may try to install a better one but likely not.

So next step would to engage the new method to delete the failed plugin?

Håkan

```
15:18:18.084   DEBUG plugin_loader.cpp:310 Skipping plugin C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\o-charts_pi.dll in safe mode
15:18:18.084 MESSAGE plugin_loader.cpp:269 Checking plugin candidate: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\ocpn_draw_pi.dll
15:18:18.084   DEBUG plugin_loader.cpp:310 Skipping plugin C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\ocpn_draw_pi.dll in safe mode
15:18:18.084 MESSAGE plugin_loader.cpp:269 Checking plugin candidate: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\polar_pi.dll
15:18:18.084   DEBUG plugin_loader.cpp:310 Skipping plugin C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\polar_pi.dll in safe mode
15:18:18.084 MESSAGE plugin_loader.cpp:269 Checking plugin candidate: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\radar_pi.dll
15:18:18.084   DEBUG plugin_loader.cpp:310 Skipping plugin C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\radar_pi.dll in safe mode
15:18:18.084 MESSAGE plugin_loader.cpp:269 Checking plugin candidate: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\vdr_pi.dll
15:18:18.084   DEBUG plugin_loader.cpp:310 Skipping plugin C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\vdr_pi.dll in safe mode
15:18:18.084 MESSAGE plugin_loader.cpp:269 Checking plugin candidate: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\watchdog_pi.dll
15:18:18.084   DEBUG plugin_loader.cpp:310 Skipping plugin C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\watchdog_pi.dll in safe mode
15:18:18.084 MESSAGE plugin_loader.cpp:269 Checking plugin candidate: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\weatherfax_pi.dll
15:18:18.084   DEBUG plugin_loader.cpp:310 Skipping plugin C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\weatherfax_pi.dll in safe mode
```
[image]
[image]

leamas, 2023-03-01:
no, we cannot make that step. Because the only way to "check" things would be to load it, an operation which fails in this case.

We have checked it in a number of ways which should find most cases. However, for this particular case  I just don't think it's possible.  

However, it might be possible to enable a manual uninstall, that's probably the best we could do. User would have to uninstall plugins one by one until opencpn starts cleanly.

leamas, 2023-03-01:
The more I think of this, I think I will make a PR of this based on this patch. This would then just make sure that -safe_mode works as intended, and looks reasonably safe to me.

Perhaps we should then let this issue be about otherwise legal plugins which generates a crash when loaded. This is by definition a very special case reflecting a plugin which has been published without being tested w r t that it's possible to load. I have a couple of ideas here, but nothing which should be designed and implemented her and now. 

One idea is to make automatic checks using opencpn-cmd before publishing. This could be done after the release, it would not require any change sin the sources. There are others...

Hakansv, 2023-03-01:
> no, we cannot make that step. Because the only way to "check" things would be to load it, an operation which fails in this case

I thought there was some "Try..." function to use when load it and if error and if a plugin we can delete it?
But that may be to look in heaven for a nice function?

leamas, 2023-03-01:
So, my suggestion is that we change the subject of this issue to something like "plugins which fails to load -> hard crash". I have the

leamas, 2023-03-01:
@rgleason ^^

leamas, 2023-03-01:
One possible route would be to create a file before loading any plugin, and delete it when it's loaded. If the load crashes, the file would be left and the plugin could be blacklisted on next restart. However, this is IMHO too much to implement at this late stage for a corner case like this.

rgleason, 2023-03-01:
I will change it, but something isn't working right!  And it has to do with what OpenCPN does when it meets a bad plugin.
OpenCPN just fails.  Try to open it and it fails, Try to open it and it fails. Try to open it and it fails. Now what do I do if I don't know why?  Ok. I guess I should uninstall OpenCPN and reinstall.  Dam that didn't work. I'll try it again.....

So I just tried to start OpenCPN, to check a route. "C:\Program Files (x86)\OpenCPN 5.7.1+71ffe26\opencpn.exe"  and it just crashed, with a nice note which tells me very little. The next time I try to start O,

[image]

So I say "Yes" and it just crashes.  So what is wrong here. ...Maybe I should reinstall OpenCPN?
No, actually the first place to go is look at opencpn.log. Ok, where is that?   Well now I've finally found it. (that's what a user is thinking after all the trouble).   It shows

```
________

13:08:57.921    INFO logger.cpp:130 ------- OpenCPN version 5.7.1+71ffe26 restarted at 2023-03-01 -------

13:08:57.921    INFO logger.cpp:130 Using loglevel INFO
13:08:57.921 MESSAGE ocpn_app.cpp:1230 wxWidgets version: wxWidgets 3.2.1 Microsoft Windows NT 64 bit wxMSW
13:08:57.921 MESSAGE ocpn_app.cpp:1235 OS Version reports as:  10.0
13:08:57.921 MESSAGE ocpn_app.cpp:1237 MemoryStatus:  mem_total: 16128 mb,  mem_initial: 21 mb
13:08:57.921 MESSAGE ocpn_app.cpp:1249 OCPN_OSDetail:  x86_64 ; msvc-wx32 ; 10 ;  ; 
13:08:57.921 MESSAGE ocpn_app.cpp:1256 SData_Locn is C:\Program Files (x86)\OpenCPN 5.7.1+71ffe26\
13:08:57.921 MESSAGE ocpn_app.cpp:1271 PrivateDataDir is C:\ProgramData\opencpn\
13:08:57.921 MESSAGE ocpn_app.cpp:1355 Using existing Config_File: C:\ProgramData\opencpn\opencpn.ini
13:08:57.937 MESSAGE navutil.cpp:1256 winPluginDir, read from ini file: 
13:08:57.937 MESSAGE navutil.cpp:1340 Setting Ownship Lat/Lon 45.7634, 13.5591
13:08:57.937 MESSAGE navutil.cpp:3198 Switch InlandEcdis mode Off
13:08:57.953 MESSAGE styles.cpp:841 Styles loading from C:\Program Files (x86)\OpenCPN 5.7.1+71ffe26\uidata\styles.xml
13:08:57.953 MESSAGE styles.cpp:818 No styles found at: C:\ProgramData\opencpn\
13:08:57.953 MESSAGE styles.cpp:801 No styles found at: C:\ProgramData\opencpn\.opencpn\
13:08:57.969 MESSAGE styles.cpp:775 Using SVG Icons
13:08:57.969 MESSAGE ocpn_app.cpp:1416 Detected display size (horizontal): 366 mm
13:08:57.969 MESSAGE OCPNPlatform.cpp:831 Adding catalog lookup path:  C:\Program Files (x86)\OpenCPN 5.7.1+71ffe26\share\locale
13:08:57.969 MESSAGE base_platform.cpp:389 Using private plugin dir: C:\Users\fcgle\AppData\Local\opencpn\plugins
13:08:57.969 MESSAGE OCPNPlatform.cpp:841 Adding catalog lookup path:  C:\Users\fcgle\AppData\Local\opencpn\share\locale
13:08:57.969 MESSAGE OCPNPlatform.cpp:882 Getting DefaultSystemLocale...
13:08:57.969 MESSAGE OCPNPlatform.cpp:903 Found LanguageInfo for: English
13:08:57.969 MESSAGE ocpn_app.cpp:1452 System default Language:  en
13:08:57.969 MESSAGE ocpn_app.cpp:1455 Config file language:  en_US
13:08:57.969 MESSAGE ocpn_app.cpp:1460 Adjusted App language:  en_US
13:08:57.969 MESSAGE OCPNPlatform.cpp:963 ChangeLocale: Language load for:  en_US
13:08:57.985 MESSAGE OCPNPlatform.cpp:1016 ChangeLocale: Locale Init OK for:  en_US
13:08:57.985 MESSAGE OCPNPlatform.cpp:1041 Loading catalog for opencpn core.
13:08:57.985 MESSAGE OCPNPlatform.cpp:882 Getting DefaultSystemLocale...
13:08:57.985 MESSAGE OCPNPlatform.cpp:903 Found LanguageInfo for: English
13:08:57.985 MESSAGE ocpn_app.cpp:1467 Opencpn language set to:  en_US
13:08:57.985 MESSAGE ConfigMgr.cpp:624 Loading Configs catalog: C:\ProgramData\opencpn\Configs\configs.xml
13:08:58.000 MESSAGE ocpn_app.cpp:1697 Creating MyFrame...size(2688, 1624)  position(-16, -16)
13:08:58.000 MESSAGE ocpn_frame.cpp:836  **** Connect stuff
13:09:18.745 MESSAGE ocpn_app.cpp:1206 

________

13:09:18.745    INFO logger.cpp:130 ------- OpenCPN version 5.7.1+71ffe26 restarted at 2023-03-01 -------

13:09:18.745    INFO logger.cpp:130 Using loglevel INFO
13:09:18.745 MESSAGE ocpn_app.cpp:1230 wxWidgets version: wxWidgets 3.2.1 Microsoft Windows NT 64 bit wxMSW
13:09:18.745 MESSAGE ocpn_app.cpp:1235 OS Version reports as:  10.0
13:09:18.745 MESSAGE ocpn_app.cpp:1237 MemoryStatus:  mem_total: 16128 mb,  mem_initial: 21 mb
13:09:18.745 MESSAGE ocpn_app.cpp:1249 OCPN_OSDetail:  x86_64 ; msvc-wx32 ; 10 ;  ; 
13:09:18.745 MESSAGE ocpn_app.cpp:1256 SData_Locn is C:\Program Files (x86)\OpenCPN 5.7.1+71ffe26\
13:09:18.745 MESSAGE ocpn_app.cpp:1271 PrivateDataDir is C:\ProgramData\opencpn\
13:09:18.745 MESSAGE ocpn_app.cpp:1355 Using existing Config_File: C:\ProgramData\opencpn\opencpn.ini
13:09:18.761 MESSAGE navutil.cpp:1256 winPluginDir, read from ini file: 
13:09:18.761 MESSAGE navutil.cpp:1340 Setting Ownship Lat/Lon 45.7634, 13.5591
13:09:18.761 MESSAGE navutil.cpp:3198 Switch InlandEcdis mode Off
13:09:18.761 MESSAGE styles.cpp:841 Styles loading from C:\Program Files (x86)\OpenCPN 5.7.1+71ffe26\uidata\styles.xml
13:09:18.761 MESSAGE styles.cpp:818 No styles found at: C:\ProgramData\opencpn\
13:09:18.761 MESSAGE styles.cpp:801 No styles found at: C:\ProgramData\opencpn\.opencpn\
13:09:18.777 MESSAGE styles.cpp:775 Using SVG Icons
13:09:18.777 MESSAGE ocpn_app.cpp:1416 Detected display size (horizontal): 366 mm
13:09:18.777 MESSAGE OCPNPlatform.cpp:831 Adding catalog lookup path:  C:\Program Files (x86)\OpenCPN 5.7.1+71ffe26\share\locale
13:09:18.777 MESSAGE base_platform.cpp:389 Using private plugin dir: C:\Users\fcgle\AppData\Local\opencpn\plugins
[... cut ...]
13:09:18.808 MESSAGE OCPNPlatform.cpp:882 Getting DefaultSystemLocale...
13:09:18.808 MESSAGE OCPNPlatform.cpp:903 Found LanguageInfo for: English
13:09:18.808 MESSAGE ocpn_app.cpp:1467 Opencpn language set to:  en_US
13:09:18.808 MESSAGE ConfigMgr.cpp:624 Loading Configs catalog: C:\ProgramData\opencpn\Configs\configs.xml
13:09:18.808 MESSAGE ocpn_app.cpp:1697 Creating MyFrame...size(2688, 1624)  position(-16, -16)
13:09:18.808 MESSAGE ocpn_frame.cpp:836  **** Connect stuff

```

So I tried deleting ocpn_draw.dll

rgleason, 2023-03-01:
Now there are no dlls in the directory and it is still crashing!!!!  Why

rgleason, 2023-03-01:
Alec I don't believe this is a "corner case" it has happened too many times to me to call it a corner case.

> One possible route would be to create a file before loading any plugin, and delete it when it's loaded. If the load crashes, the file would be left and the plugin could be blacklisted on next restart. However, this is IMHO too much to implement at this late stage for a corner case like this.

leamas, 2023-03-01:
Well, we are all developers and subject to corner cases not affecting regular users.

> Now there are no dlls in the directory and it is still crashing!!!! Why

No idea, and a log from a session without `--loglevel debug` does not tell that much. Please re-run using that option and attach the new logfile

bdbcat, 2023-03-01:
Rick...
I observe that the crash described by VS screenshots above is happening in o-draw_pi.dll.  Not in OCPN core.  As Hakan mentioned, o-draw seems to be returning a faulty bitmap pointer. This information is clear in the text associated with the crash.

Ergo, this is a "bad" plugin.  We must assume that plugins are tested by devs and Beta testers before publication, so that the chances of a **user** seeing a faulty plugin crash like this may be minimized.  But we can certainly expect to see this sort of crash in Alpha/Beta testing of plugins.

My advice:  focus on the individual plugin, not the PIM, as first most likely source of problem.  After all, several other plugins work just fine with the O571 PIM.

closed 2023-03-01 by bdbcat

rgleason, 2023-03-02:
But Dave, if the plugin is not compatible and "BAD" isn't it possible to protect OpenCPN from a crash?  Or is this very difficult to do programmatically?

BTW Jon seems to think this is an OpenCPN problem. I've made an Issue under Ocpn_Draw (see one of the links above) .

Right now I can't seem to fix the instant bomb outs when using Beta 4. I first removed ocpn_draw.dll and that did not fix the problem. So perhaps it is unrelated.

Also I have removed all plugin dll from  C:\Users\fcgle\AppData\Local\opencpn\plugins
There are still  vcruntime140.dll, SglW32.dll, portaudio_x86.dll, oexservrd.exe and many plugin subdirectories.

bdbcat, 2023-03-02:
Rick...
We cannot protect O against all the possible ways that a third-party plugin may fail.  What we _**can**_ do is to curate the catalog, so that average users will not see these types of frank crashes on plugi load..  And we "test them away".

"Jon seems to think this is an OpenCPN problem."
OCPN is crashing while executing code within the o-draw.dll file.  Seems plain to me that o-draw is the location of the problem, or at least that would be my working assumption.
How can I help you and Jon correct whatever is the fault in o-draw?  Do you have a stable build environment running on Windows?

"Right now I can't seem to fix the instant bomb outs when using Beta 4. I first removed ocpn_draw.dll and that did not fix the problem. So perhaps it is unrelated."
Probably unrelated.  I could help on Anydesk when convenient.

rgleason, 2023-03-02:
starting from command prompt in the program files (x86) / opencpn + xxxx directory
opencpn --loglevel debug    starts opencpn and it seems to work ok.
First I updated master catalog and  loaded one of the recent updated TP plugins built using 1.0.226 Template  (Climatology)  It loaded and worked properly.  Then I tried to Import Plugin  the ocpn_draw plugin  (version x 21)  that I have uploaded above.  

I got a message that the plugin  is compatible!!!!!!!!!!!!!!!   ---When it is NOT.

[image]

When I installed the plugin after loading, and prior to Enabling, Opencpn bombed

Attached is a new log file.

PM is not capable of finding bad code, it is just able to see if a plugin is "intended for a given OS.
 
[opencpn.log.txt](https://github.com/OpenCPN/OpenCPN/files/10866572/opencpn.log.txt)

bdbcat, 2023-03-02:
"PM is not capable of finding bad code, it is just able to see if a plugin is "intended for a given OS."
Absolutely correct statement.
PM cannot predict whether a plugin will crash or not.  No-one can, reliably.
I re-iterate that plugins that crash on load should not be in the catalog in the first place.  We prevent that by testing.
That is what we are doing here, trying to verify that that policy is being followed.  And we have a failure, so let us fix the plugin.

bdbcat, 2023-03-02:
"Perhaps we should have some instructions in the manual about this problem."
Our instructions should be, in 3 steps:
1.  Restart in safe mode.
2.  Using PM, delete the last known plugin installed.
3.  Restart O, and contact  the plugin author/maintainer for assistance.

Yes, yes, I know that this process does not work on Beta#4. But I just pushed a PR from Alec that should fix that problem.

rgleason, 2023-03-02:
After opening OpenCPN using the command prompt opencpn  --loglevel debug
The program appears to open normally, but as soon as I go to Options > Plugins  it bombs out STILL!!!

Jon's cunningly BAD plugin has totally disabled OpenCPN so that it does not work!!
What is causing this problem?
What should a developer do the clear a cache or whatever?
Is this happening because the plugin cache is corrupted?
Should that be deleted now?

[opencpn.log.txt](https://github.com/OpenCPN/OpenCPN/files/10866662/opencpn.log.txt)

bdbcat, 2023-03-02:
We cannot tell exactly where O is failing form the log.
We will need to do a detailed debug session in VS to find the trouble.  Anydesk?

rgleason, 2023-03-02:
This is impossible to do

> Using PM, delete the last known plugin installed.

OpenCPN fails when Options > Plugins is selected!!!!!!
...and it keeps failing.

OPenCPN should automatically clear the cache on startup when this happens.

rgleason, 2023-03-02:
OK anydesk.

bdbcat, 2023-03-02:
We have an unrelated bug that is crashing your system.  Let's fix that, please?

rgleason, 2023-03-02:
Thanks. Dave.

bdbcat, 2023-03-02:
Close?

PR #3040 (2023-03-01): plugin_loader et. al.: Make sure safe_mode works as intended (#3038).
-safe_mode just does not work as intended. Fix that.

This does not close #3038, but streamlines the remaining issue.