OpenCPN/OpenCPN #3041: Under OCPN5.7.1 beta 4 sub-windows are not attached to a parent canvas/frame

antipole2, 2023-03-01:
Windows and wxDialogs are often created as sub-windows of a parent. A plugin can do this via the API GetOCPNCanvasWindow().
Thus, if the frame is dragged around the screen, a non-modal dialogue follows the frame and keeps its position relative to the frame.  At least that is how it works with OXPN5.6.2/wx3.1.5.

**Describe the bug**
Under OCPN5.7.1 beta 4 dialogues do not move with the frame.

**To Reproduce**
Open WMM from the tool bar. 
Drag the frame around and the WMM window does not follow it.

**Expected behavior**
The window should follow the frame.

A further problem is that, as the dialogue appears not to be a sub-window, when focus moves to the main window, it covers the dialogue window.  This does not happen under OCPN5.6.2.

**Desktop (please complete the following information if applicable):**
 - OS: MacOS
 - Version Ventura.
 - Mike has confirmed he sees this under Windows too.

**Additional context**
This behaviour could be explained if  `GetOCPNCanvasWindow()` was now returning `nullptr`.
We have checked and it is returning a likely pointer value.

Not sure if this is an OpenCPN 5.7.1 issue or a change in the move from wxWidgets 3.1 to 3.2.

nohal, 2023-03-01:
As already discussed on Zulip, this is a correct behavior under macOS (Every application on my system, be from Apple or other developer, behaves like this). Non-modal windows submerge when they don't have focus unless they use `wxFLOAT_ON_PARENT` style and don't get moved with the parent window. wxWidgets 3.2 simply fixed the incorrect behavior of the older versions here.

antipole2, 2023-03-02:
OK - let's close this one.  I am about to raise a separate issue for multiple screens/spaces.