OpenCPN/OpenCPN #3047: macOS Monterey 12.6 with OpenCPN 5.7.2 will only accept non-wx32 plugin which then fails to run

jongough, 2023-03-03:
**Describe the bug**
Running macOS Moneterey 12.6 under qemu/kvm on Ubuntu 22.04 fails to load and run wx32 plugin reporting incompatible plugin. It will load a non-wx32 plugin which then fails as a symbol cannot be found

**To Reproduce**
Steps to reproduce the behavior:
1. Install OCPN 5.7.2 from beta download location
2. Run OCPN
3. Install OCPN 1.8.23.0, non wx32 version as wx32 version is 'not compatible'
4. Plugin fails to instantiate with a symbol not found error.

**Expected behavior**
The wx32 version of the plugin should install and run

**Screenshots**
N/A

**Desktop (please complete the following information if applicable):**
 - OS: Ubuntu desktop, macOS 12.6 guest under qemu/kvm
 - Version 22.04

Log file from run
```
16:56:30.186    INFO logger.cpp:130 ------- OpenCPN version 5.7.1+c5ae2a0 restarted at 2023-03-03 -------

16:56:30.186    INFO logger.cpp:130 Using loglevel INFO
16:56:30.186 MESSAGE ocpn_app.cpp:1228 wxWidgets version: wxWidgets 3.2.2.1 Apple Mac OS X 64 bit wxMac
16:56:30.186 MESSAGE ocpn_app.cpp:1233 OS Version reports as:  12.6
16:56:30.186 MESSAGE ocpn_app.cpp:1235 MemoryStatus:  mem_total: 4096 mb,  mem_initial: 91 mb
16:56:30.186 MESSAGE ocpn_app.cpp:1247 OCPN_OSDetail:  x86_64 ; darwin-wx32 ; 10.15.7 ;  ;
16:56:30.186 MESSAGE ocpn_app.cpp:1254 SData_Locn is /Applications/OpenCPN.app/Contents/SharedSupport/
16:56:30.186 MESSAGE ocpn_app.cpp:1269 PrivateDataDir is /Users/jon/Library/Preferences/opencpn
16:56:30.186 MESSAGE ocpn_app.cpp:1353 Using existing Config_File: /Users/jon/Library/Preferences/opencpn/opencpn.ini
16:56:30.237 MESSAGE navutil.cpp:1256 winPluginDir, read from ini file:
16:56:30.237 MESSAGE navutil.cpp:1340 Setting Ownship Lat/Lon 33.358, -79.282

---------------------------------------------------

17:02:09.993 MESSAGE pluginmanager.cpp:4683 Copied /Users/jon/Downloads/ocpn_draw_pi-1.8.23.0-darwin-x86_64-11.4-macos.tar to local cache
17:02:09.994 MESSAGE plugin_loader.cpp:243 PluginLoader: loading plugins from /Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns;/Applications/OpenCPN.app/Contents/Plugins
17:02:09.994 MESSAGE plugin_loader.cpp:178 Using DYLD_LIBRARY_PATH: /Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns:/Applications/OpenCPN.app/Contents/Plugins:/Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns:/Applications/OpenCPN.app/Contents/Plugins:/Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns:/Applications/OpenCPN.app/Contents/Plugins
17:02:09.994 MESSAGE plugin_loader.cpp:193 Using PATH: /Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns:/Applications/OpenCPN.app/Contents/Plugins:/Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns:/Applications/OpenCPN.app/Contents/Plugins:/Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns:/Applications/OpenCPN.app/Contents/Plugins:/usr/bin:/bin:/usr/sbin:/sbin
17:02:09.994 MESSAGE plugin_loader.cpp:248 Loading plugins from dir: /Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns
17:02:09.995 MESSAGE plugin_loader.cpp:420 PluginLoader searching for PlugIns in location /Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns
17:02:09.995 MESSAGE plugin_loader.cpp:458 Found 1 candidates
17:02:09.995 MESSAGE plugin_loader.cpp:268 Checking plugin candidate: /Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns/libocpn_draw_pi.dylib
17:02:09.995 MESSAGE plugin_loader.cpp:1171 Plugin is compatible: true
17:02:09.995 MESSAGE plugin_loader.cpp:1186 PluginLoader: Loading PlugIn: /Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns/libocpn_draw_pi.dylib
17:02:09.996   ERROR dlunix.cpp:120 Failed to load shared library '/Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns/libocpn_draw_pi.dylib': dlopen(/Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns/libocpn_draw_pi.dylib, 0x0002): symbol not found in flat namespace (__ZN11wxAnyButton11DoSetBitmapERK8wxBitmapN15wxAnyButtonBase5StateE)
17:02:09.996 MESSAGE plugin_loader.cpp:1241    PluginLoader: Cannot load library: /Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns/libocpn_draw_pi.dylib
```

Note OCPN claims to be running wx 3.2.2.1, plugin that is accepted is running wx 3.1.5 and plugin that is rejected as not compatible is running wx 3.2.1.

ocpn_draw_pi is available for testing from cloudsmith opencpn ocpn_draw_pi-beta repository.

leamas, 2023-03-03:
Just to gather some more info... If you open the Plugins  | Settings  dialog after the usual CatalogExpert=1:

  - What is the active compatibility  reported?
  - If you change the compatibility to *darwin-wx-32*, does then everything work as it should?

bdbcat, 2023-03-03:
Tested on real Mac, two devices.
1.  Using current OCPN core build, from Cloudsmith, [https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.7.1.pkg](url)
2.  O reports flavor "darwin-wx32"
3. Using current Beta plugin catalog.
4. o-draw version 1.8.21.1 is available in the catalog, installs and loads without error.
5. Some functional issues.  Mainly, the "Draw toolbar" flashes once and disappears.  If it helps, left-click **held down** on main toolbar o-draw icon keeps the draw toolbar on screen.  So maybe some "button-up" processing going astray.

tl;dr;
o-draw 1.8.21.1 loads and starts correctly on current O build, real hardware.

bdbcat, 2023-03-03:
More...
imported o-draw 1.8.23.0 by direct load of tarball from Cloudsmith.
Identical results as above.

jongough, 2023-03-03:
This is what is shown when looking at Plugin Catalog Settings:
[image]

bdbcat, 2023-03-03:
Did you build the OCPN core locally, or use the current Cloudsmith image?

jongough, 2023-03-03:
I used the one found in the beta forum stream. I downloaded the package and installed it with no changes.
I have just downloaded the one you pointed to above and installed it and it works, but I may have to delete the ini file to get a 'clean' install.

The log of the run if the install just now:
```
06:20:21.302        INFO logger.cpp:130 ------- OpenCPN version 5.7.1+50a020f restarted at 2023-03-04 -------

06:20:21.302        INFO logger.cpp:130 Using loglevel INFO
06:20:21.303 MESSAGE ocpn_app.cpp:1228 wxWidgets version: wxWidgets 3.2.2.1 Apple Mac OS X 64 bit wxMac
06:20:21.303 MESSAGE ocpn_app.cpp:1233 OS Version reports as:    12.6
06:20:21.303 MESSAGE ocpn_app.cpp:1235 MemoryStatus:    mem_total: 4096 mb,    mem_initial: 76 mb
06:20:21.303 MESSAGE ocpn_app.cpp:1247 OCPN_OSDetail:    x86_64 ; darwin-wx32 ; 10.15.7 ;    ; 
06:20:21.303 MESSAGE ocpn_app.cpp:1254 SData_Locn is /Applications/OpenCPN.app/Contents/SharedSupport/
06:20:21.303 MESSAGE ocpn_app.cpp:1269 PrivateDataDir is /Users/jon/Library/Preferences/opencpn
06:20:21.303 MESSAGE ocpn_app.cpp:1353 Using existing Config_File: /Users/jon/Library/Preferences/opencpn/opencpn.ini
06:20:21.307 MESSAGE navutil.cpp:1256 winPluginDir, read from ini file: 
06:20:21.307 MESSAGE navutil.cpp:1340 Setting Ownship Lat/Lon 33.358, -79.282
```
This version is `OpenCPN version 5.7.1+50a020f`, the previous one was `OpenCPN version 5.7.1+c5ae2a0`. Not sure what changed.

jongough, 2023-03-03:
The toolbar is a Z-order issue which I think has happened before where OD becomes the top window leaving the toolbar behind. The icon size does not appear to change when set in the OD preferences as it does on other platforms. This 'may' be due to having to create the icons for OCPN prior to the plugin knowing what its own settings are. These are found in the `init` function.

bdbcat, 2023-03-03:
Toolbar z-order:
If I minimize the main OCPN frame, I do not see the toolbar underneath, as I expected to.
Thoughts?

On the icon size:  Since you create a struct to hold all necessary icon pointers, most of the struct contents could be deferred until Init().  Only the "main" icon, used for OCPN setting PIM list, need be available before Init().  So, a two stage icon build strategy.

jongough, 2023-03-03:
Try moving OCPN to one side and you will/should see the toolbar. However, I have just tried minimising OCPN and the toolbar is still there. I wonder if this is another 'artifact' of virtualisation' or if there is another problem?

I was thinking about doing most of the icon work at init rather than instantiation. I will see if I can make that work. What does OCPN need the icons for at load time rather than init time? Is it the plugin manager screen or something else?

bdbcat, 2023-03-03:
I'll look for the toolbar again.

Icons:
" Is it the plugin manager screen?"  Yes, exactly. 
The method   " virtual wxBitmap *GetPlugInBitmap();" will be called very soon after instantiation, even if Init() is not called.  This method should return the simple "identifier" bitmap associated with the plugin.

rgleason, 2023-03-16:
JG is overseas, may not see this.

nohal, 2023-04-15:
I believe we have a functional system in place.

closed 2023-04-15 by nohal