OpenCPN/OpenCPN #3053: API1.18 WriteCommDriver does not work

antipole2, 2023-03-06:
**Describe the bug**
WriteCommDriver API has not been implemented (as at 6th March 2023)

**To Reproduce**
Steps to reproduce the behavior:
Perform a global text search on the src for OpenCPN for 'WriteCommDriver'
This finds `CommDriverResult WriteCommDriverN2K(` but not `CommDriverResult WriteCommDriver(`

**Expected behavior**
Expect to be able to find it.

bdbcat, 2023-03-07:
Write CommDriver implemented in github master now.

For reference, here is my (unoptimized, but instructive) test code for this interface.

```
      //  Test WriteCommDriver
      // Send a string to comm port "/dev/ttyUSB0"

      std::string payload_msg("$GPRMC, some data \r\n");
      std::string my_port("/dev/ttyUSB0");

      // Find the N0183 driver at specified port

      // Get a list of active drivers
      const std::vector<DriverHandle> driver_handles = GetActiveDrivers();

      std::string my_driver_handle;

      // Find an N0183 driver at specified port
      for (unsigned int i=0 ; i < driver_handles.size(); i++) {
        const std::unordered_map<std::string, std::string> &attributes = GetAttributes(driver_handles[i]);
        bool bProtocol_OK = false;
        bool bPort_OK = false;
        for (auto it = attributes.begin(); it != attributes.end(); it++) {
          std::string attrib_name = it->first;
          std::string attrib_value = it->second;
          if (attrib_name.find("protocol") != std::string::npos) {
            if (attrib_value.find("nmea0183") != std::string::npos) {
              bProtocol_OK = true;
              break;
            }
          }
        }
        if (bProtocol_OK) {
          for (auto it = attributes.begin(); it != attributes.end(); it++) {
            std::string attrib_name = it->first;
            std::string attrib_value = it->second;
            if (attrib_name.find("commPort") != std::string::npos) {
              if (attrib_value.find(my_port) != std::string::npos) {
                bPort_OK = true;
                break;
              }
            }
          }
        }

        if (bProtocol_OK && bPort_OK) {
          my_driver_handle = driver_handles[i];
          break;
        }
      }

      if( my_driver_handle.size()) {
        // Prepare the payload
        std::vector<uint8_t> payload_buffer;
        for (unsigned int i=0; i < payload_msg.size(); i++)
          payload_buffer.push_back(payload_msg[i]);

        auto payload = std::make_shared<std::vector<uint8_t>>(payload_buffer);

        //perform the write operation
        CommDriverResult result = WriteCommDriver( my_driver_handle, payload);
     }

```

leamas, 2023-03-07:
Or, using some C++11:

```
// Test WriteCommDriver, send a string to comm port "/dev/ttyUSB0"
// Compile (not link): gcc $(wx-config --cflags) -I$PWD/include main.cpp

#include "ocpn_plugin.h"

int main(int argc, char** argv) {
  using namespace std;
  string payload_msg("$GPRMC, some data \r\n");
  string my_port("/dev/ttyUSB0");

  for (const auto& handle : GetActiveDrivers()) {
    const auto& attributes = GetAttributes(handle);
    if (attributes.find("protocol") == attributes.end()) continue;
    if (attributes.at("protocol") != "nmea0183") continue;
    if (attributes.find("commPort") == attributes.end()) continue;
    if (attributes.at("commPort").find(my_port) == string::npos) continue;

    auto payload = make_shared<std::vector<uint8_t>>();
    for (const auto& ch : payload_msg) payload->push_back(ch);
    CommDriverResult result = WriteCommDriver(handle, payload);
    break;
  }
}

```

antipole2, 2023-03-07:
Thanks both and for the code examples - most useful.

antipole2, 2023-03-07:
Sure  - Have not got to testing it myself yet but bdbcat has, so let's close it.

closed 2023-03-07 by antipole2